SynthstromAudible/DelugeFirmware #800: Incremental per-note sequencing does not work

mpancia, 2023-12-08:
### Please describe the problem:

On a fresh install of the nightly firmware `deluge-v1_0_0-nightly+2023_12_04-2124e59`, attempting to set a per-note probability that happens per-interval (by turning clockwise on the `SELECT` knob) produces a box that says `T` with no other text.

### What is the expected behavior?

Turning the `SELECT` knob clockwise scrolls through interval options for the note.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_12_04-2124e59

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turned on.
2. Created a new synth clip with a new synth.
3. Placed a C note on the first step.
4. Held down the note and twisted the select knob clockwise. 
5. Display overlays an empty box at the first turn, then at the second turn displays a box with just "T". Further turns do not change the display.

SchluppiBobbz, 2023-12-09:
I can confirm the bug on my Deluge with 4SEG. However, nothing is displayed there, not even a T ;).

soymonitus, 2023-12-14:
Confirmed to happen also on setting row probability

PR #836 (2023-12-16): Fix iterations not displaying
displayString variable was unnecesary and not set in the case of iteration when optimizations are on

removed it and just pass buffer to popup instead

Fix #800
closed 2023-12-16 by bobtwinkles