SynthstromAudible/DelugeFirmware #804: More Dropouts Occurring with C1_0_0 Firmware

Tsapio, 2023-12-11:
### Please describe the problem:

Love it all the new features. Grid mode I now can’t live without. But I’m experiencing a lot more dropouts of tracks sooner with the new version. 
min version 4.1.3 I could play a specific song with 9 tracks. Now I can’t even play 6 tracks on the same song without dropouts occurring. 

### What is the expected behavior?

All things being equal except the new firmware update, dropouts should not be occurring with so few tracks playing. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

C1.0.0

### If possible provide the steps to reproduce the issue and upload additional media:

I reverted back to firmware 4.1.3 and played up to 9 tracks without dropouts. I them updated to firmware C1.0.0 and played the song with the same tracks active which caused numerous dropouts. I then began removing tracks and could only get up to 6 tracks to play at a time without dropouts.

m-m-adams, 2023-12-11:
Please provide the song saved with collect all - we can't do anything to help otherwise

m-m-adams, 2023-12-11:
Awesome thanks!

m-m-adams, 2023-12-11:
FYSA this is unlikely to have a quick fix - this song is performing worse than it should (you get 10 voices, should normally be 15ish sample voices) but it's not immediately clear why

Thanks for the report and we'll follow up once we have something to test!

Tsapio, 2023-12-12:
Thank you! Although I’ve never gotten anywhere close to 15 voices. That would be amazing! On the old firmware I got maybe to 9 but that was stretching it and not always. 

With this particular song though it doesn’t drop on the older firmware. Just on the C1 firmware. So thought I’d share since it seemed like a performance issue linked to the C1 firmware. 

Even now when I switch back to 4.1.3 it doesn’t drop. 

Thank you so much for looking into it!

m-m-adams, 2023-12-12:
It'll depend on exactly what's in the voices but 15 sample voices playing is loosely what we'd expect. If you just use FM synths you can get around 100 voices! 

This song is getting about 10 right now at its peaks which should be manageable, so it's likely there's a state that's being mishandled

Tsapio, 2023-12-12:
@m-m-adams thank you.  By "mishandled" do you mean something on my end or by the machine itself.  Do the dropouts not occur in the new firmware for you?

Tsapio, 2023-12-12:
To add to this because it’s another issue in the same song file…

In Grid mode, when the red section is selected, if I select yellow, red doesn’t mute. If any other color is selected, red will mute. But selecting yellow when red is selected already does not mute red. Are you able to reproduce this by chance with the song file?

https://github.com/SynthstromAudible/DelugeFirmware/assets/153487424/925af115-b3c4-4faa-abe8-8d5300bb8351

m-m-adams, 2023-12-12:
Same for me on the song file

For performance the key problem is loading the samples for the blue kit in the yellow row

If you humanize that clip so that the 6 notes don't start at once the culling is smooth and less audible

m-m-adams, 2023-12-12:
> @m-m-adams thank you. By "mishandled" do you mean something on my end or by the machine itself. Do the dropouts not occur in the new firmware for you?

No I meant in the firmware! Like I think this song should be fine, we'll just need to think a bit about prioritizing and scheduling loads

Tsapio, 2023-12-12:
Hmmm I tried quantizing the kit but no luck. Tried both quantizing an individual sound (the snare dropping is the biggest issue) and the whole thing. 
I tried muting everything on the yellow row and then just adding back the kit, then synths only, then as I started adding audio clips it happened. But the main culprit seems to be the light blue Audio5 clip. The moment I add that one the snare clips even if the others are muted.

Tsapio, 2023-12-13:
@m-m-adams yeah at this point I'm out of things to try.  Turning off Grain, Humanizing, nothing allows as many voices as when I switch back to 4.1.3.  I'm not going to switch back long term because this update is just glorious but it really does limit what I can do with it when I can only play up to 4-5 tracks at a time.
I know you said this may be a longer term fix so I'll be patient.  Just want to let you know I've tried everything so far.  
Again, absolutely loving all the hard work.  It's a whole new machine!

m-m-adams, 2023-12-13:
Out of curiosity what SD card are you using? I can see if I can find my original one to test but I generally use a faster card

Tsapio, 2023-12-14:
Yeah I'm using the SD card that came with the Deluge.  The stock one.

Tsapio, 2023-12-19:
@m-m-adams I know there's a ton happening and you have probably bigger things you've got going on.  Just checking in on any update with the performance issues with the new Grid Mode.  It seems to be happening across the board for me where I can only play about 4, maybe 5 tracks and then start getting drops.  It kind of makes Grid Mode unplayable almost unfortunately.  Thanks again.

Tsapio, 2024-01-05:
@songsta well I'm really sorry to hear that cuz it sucks but I'm glad to know I'm not the only one.  I've pretty much had to go back to the stock firmware too.  I hate it because I really like grid view for performance and the grain effect and automation, etc.  It's a brilliant update.  I just need more voices as well.  Thanks for letting me know man! I'll watch that ticket too.

solaris76, 2024-01-11:
Just want to add I'm having this issue too.

ok-reza, 2024-01-15:
> To add to this because it’s another issue in the same song file…
> 
> In Grid mode, when the red section is selected, if I select yellow, red doesn’t mute. If any other color is selected, red will mute. But selecting yellow when red is selected already does not mute red. Are you able to reproduce this by chance with the song file?
> 
>  72404593221__F031CA8E-E1AB-4481-B08D-E9D3A7FA7AE5.MOV

this is a user error: you've set the yellow section to 'Launch Non-Exclusively', if you don't want that behavior then switch to blue mode, press and hold that row's section color and turn clockwise once.

Tsapio, 2024-01-17:
@ok-reza thank you for that fix!  I didn't realize that was even an option. Not sure how I did that in the first place.

chewabledrapery, 2024-01-20:
As #941 was closed i'm resharing this project, not sure if i needed to do this so please forgive me if i'm being stooooopid :)

https://github.com/SynthstromAudible/DelugeFirmware/files/13990726/Example1.2.zip

Thanks all

m-m-adams, 2024-01-23:
Can you try this build - https://github.com/SynthstromAudible/DelugeFirmware/pull/1036? It seems to play "I'll make you a deal" from @Tsapio without noticeable drops

More improvements to come!

m-m-adams, 2024-01-23:
We've merged that to community so checkout the nightly build - @chewabledrapery as far as I can tell it makes your song play properly as well

chewabledrapery, 2024-01-23:
> We've merged that to community so checkout the nightly build - @chewabledrapery as far as I can tell it makes your song play properly as well

Ahh awesome thanks for the update, was just trying to figure out how to get the bin, is that on the nightly now or scheduled for later today? Not sure when the nightlys are compiled/uploaded? 

Thanks again I'll try it out and report back!

chewabledrapery, 2024-01-23:
@m-m-adams I have flashed deluge-v1_1_0-nightly+2024_01_23-8fea0f5.bin 

The issue does seem to be reolved in the example project that i provided, but still persists in other projects, I will prepare another example that exhibits the behaviour and upload.

Many thanks! Appreciate all the effort :)

Evilpilli, 2024-02-02:
I've had this issue earlier, and can at least report that with a new SD card, sdxc, it was partly mitigated. And on the newest nightly it seems to be fixed entirely. My older songs are now playing ad expected.

Tsapio, 2024-02-04:
@chewabledrapery & @m-m-adams I have a song file for you with dropouts to test but when I try to save it gives me an error, doesn't allow it and says to report the bug.
[image]

m-m-adams, 2024-02-05:
how are you trying to save it? have you been able to save without collect all?

Tsapio, 2024-02-06:
Actually that helped.  I just retyped the name.  Thanks!
[Wake Up_2.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14173384/Wake.Up_2.zip)

m-m-adams, 2024-02-13:
If you could test this PR it would be appreciated - https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/7893732206

I believe it's better, it culls less but also does it differently so I'm not 100% sure

Tsapio, 2024-02-14:
> If you could test this PR it would be appreciated - #1225
> 
> I believe it's better, it culls less but also does it differently so I'm not 100% sure

@m-m-adams just making sure, you aren't addressing me right?  Sorry.

Tsapio, 2024-02-15:
> @Tsapio should have clearer - this PR adds around 4 voices to the last song you shared (wake up 2), and makes all the culls fade out instead of clicking

Omg that is be amazing!! I just didn’t know if it was a firmware version to download and test somewhere. Or just an update coming soon. That’s fantastic!

closed 2024-02-17 by m-m-adams

m-m-adams, 2026-10-26:
Hey so several years later this song no longer culls at all :)

chewabledrapery, 2026-10-26:
Awesome, really looking forward to the next cfw stable release!

PR #1225 (2024-02-17): Voice efficiency
Three changes to increase voice count - 
1. <s>When a sample is no longer needed, add it to the start of the stealable queue instead of the end</s> Removed - negative side effects with songs close to max load but not over it
2. After rendering the song, always try rendering it again (exits if it's ahead or after 5 renders). Previously only done when clocks interfered with rendering
3. Re introduce the cull smoothing - in combo with 2 this no longer has the risk of over shooting our samples, as if it's not keeping up with output it'll go again and cull a second time immediately [... cut]