SynthstromAudible/DelugeFirmware #813: LFO2 has a sync pad that's not (yet?) implemented

m-m-adams, 2023-12-12:
### Please describe the problem:

LFO1 has both rate and sync

LFO2 only has rate

LFO2 should also have a rate, to allow syncing to 1/4 or 1/8 notes while still being voice specific and reset by note-ons

### What is the expected behavior?

LFO2 should also have sync

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

nikodemus, 2024-05-12:
I'm still missing a lot of details, but I have a rough idea how to get this done nicely.

1. Move all LFO parameters to LFOConfig object or similar. Sound has then globalLFO, globalLFOConfig, and localLFOConfig. Voice has lfo (aka local LFO) just as before.

2. Put a pointer to the LFOConfig to the LFO itself: for sound just link globalLFO to globalLFOConfig when Sound() is initialized, for voice link/unlink to the correct sound when the voice is assigned/unassigned.

3. Move timeStartedSkippingRenderingLFO to the LFO object.

4. At this point resyncGlobalLFO() can become LFO::resync(), and setLFOGlobalSyncType/Level/WaveType can become LFO::setSyncType/Level/Wavetype. These methods will then be used to modify local LFO as well.

5. Add LFO2 sync stuff to the UI, hook it up just like for LFO1 pretty much.

---

So. Does this sound about right, or did someone envision a nicer way to do this?

I'm not super happy about linking LFOConfig as described above, but can't really see a way to avoid it while making code cleaner: otherwise we just end up with more semi-duplicated parameters in more places.

sapphire-arches, 2024-05-12:
for 2, you should be able to walk the existing `assignedToSound` pointer to find the LFOConfig for the local LFO. Otherwise, seems good to me!

I'm not sure I understand what `timeStartedSkippingRenderingLFO` is used for, but it does sound like it should live on the LFO.

nikodemus, 2024-05-12:
Re. 2, pointer walk means global and local LFO would work differently: I'm trying to organize things so that they could share the logic. 

However, since LFO2 retriggers, it's sync is going to be different anyhow... I'll need to think a bit more.

PR #2005 (2024-05-24): LFO: sync support for LFO2
- In addition to the UX hooks and strings, the only needed change
  is the phase computation, which we can share with LFO1 code.

- Unlike LFO1, LFO2 has no resync. Assumption is that there isn't
  significant drift during a single note, or that in case of a drone
  any drift is hopefully interesting rather than unwelcome.

- Closes #813.
closed 2024-05-24 by sapphire-arches