SynthstromAudible/DelugeFirmware #816: Voice stealing with sample-heavy song

songsta, 2023-12-13:
### Please describe the problem:

Playing a sample-heavy song on the official firmware doesn't lead to voice stealing, but immediately happens on community. 

To illustrate, I've attached the song. It will start at the correct section that shows the voice stealing. Track 6 is a kit with many different samples loaded. On play you will hear that the first sample cuts out when the second sample comes in. 

Also, track 3 is a Boards of Deluge Sample-based patch, with 2 notes held. You will hear that sometimes it plays 1 note, then the other, or sometimes both.

### What is the expected behavior?

Playing the same track on the official firmware doesn't show the same issues.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

The dropbox file can be found here.

https://www.dropbox.com/scl/fo/ca2oskcc89gatrms5tzcl/h?rlkey=wukdlna2ueu9nzear55beqyef&dl=0

Tsapio, 2024-01-05:
Just going to 2nd this with [my issue previously raised ](https://github.com/SynthstromAudible/DelugeFirmware/issues/804)which I think the team is aware.  Just a reminder so I don't feel like I'm the outlier. :)
Thanks guys.  :)

m-m-adams, 2024-01-19:
Closing as this is the same as #804 and I want to consolidate the issue

You're not forgotten, it's just hard!

closed 2024-01-19 by m-m-adams