SynthstromAudible/DelugeFirmware #818: Re-arming fill while in progress sends it into infinite loop.

hexagon5un, 2023-12-13:
### Please describe the problem:

"Disarming a fill clip when playing will stop immediately." seems not to work. Instead, if you re-arm it while the fill is playing, it loops the fill nonstop. This can be triggered by the mute/launch button, by the selection button that the fill belongs to, or in grid mode.

### What is the expected behavior?

Fill should cancel immediately.

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/pull/196

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

Amadeus 1.0

### If possible provide the steps to reproduce the issue and upload additional media:

Make a 4-bar loop
Make a 1-bar fill
Arm fill -- it plays in 4th bar as it should
Arm fill again while still playing -- the fill goes into infinite (1-bar) loop.
Arm again makes it stop, but only if you use the mute/launch or grid button -- doing it via section-select in either row or grid mode doesn't stop it.

0beron, 2023-12-13:
Thanks for the bug report, glad the feature is seeing some use!
I agree it would make sense if fill clips could never get stuck looping - I'll get to this hopefully soon.

hexagon5un, 2023-12-13:
Awesome.  And let me know if there's any way I can help out.

  Elliot.

Quoting 0beron - ***@***.*** (2023-12-13 14:12:02)

0beron, 2023-12-22:
Found the cause of this one - pressing the status pad while a fill is playing will now insta-stop it. For the case where a fill clip is activated while playback is stopped, what should happen once playback is started?
There are some possibilities:
1 - the fill will not play, and will deactivate at start of playback.
2 - the fill will deactivate but arm at start of playback. According to the longest clip that is also playing, it will play once and then stop.

Also there is the question of what to do when playback stops - currently if a fill clip is playing, it stays activated.
In case 1 above this doesn't matter as it will stop once playback resumes. In case 2 I think stopping playback should stop all fill clips so they don't play again at the start, but you could manually arm them before playback if you wanted.

soymonitus, 2024-01-26:
Now that we also hace "Once" clips, which also suffer from the same issue in which if they are active before pressing play, they will play forever the same as Fill clips. I would suggest to fix both that, if they are enabled before pressing play, when you press play:
- Once clips: they start playing right away but are armed to stop after that.
- Fill clips: they are disabled initially when play is pressed, but armed to start at a later moment.

Regarding the other issue, in which if you press them when they are playing they are unarmed to stop, so they play forever. I suggest that when you press them while playing, that they are instantly stopped, just the same as if you did Shift + press

m-m-adams, 2024-03-03:
Closing - easy solution is "don't arm them while stopped" for now, we can reopen a new issue if one of you is going to work on this

closed 2024-03-03 by m-m-adams