SynthstromAudible/DelugeFirmware #819: Changing clip from default to fill doesn't work in grid mode

hexagon5un, 2023-12-13:
### Please describe the problem:

In grid mode, pressing the button corresponding to a clip and pressing the select knob doesn't bring up the default/fill menu.

### What is the expected behavior?

I expected that, since the grid button is the equivalent of mute/launch in rows mode, that I would be able to convert a clip to a fill there too.

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/pull/196

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

Amadeus 1.0

### If possible provide the steps to reproduce the issue and upload additional media:

In rows mode: 

Hold mute/launch and press the Select knob in -- you get a default/fill choice.  Woot!

In grid mode: 

Hold the grid button corresponding to the clip that you want to convert to a fill and press the Select knob.  Nothing happens.

0beron, 2023-12-13:
Will take a look at this soon, thanks for the report.

m-m-adams, 2024-03-03:
I think this will need a third colour for grid to manage the clip itself - currently blue is the equivalent of pressing a row button, green is the equivalent of mute launch, and we don't have an equivalent for audition/section

0beron, 2024-03-04:
The default/fill/once menu is currently on the mute/launch pads in rows mode. The audition / section pads are present in grid mode in the sidebar already, I think you can edit their repeat count from there.
The behavior of the rows mode menu could be brought into the green grid mode and come up when you hold a grid pad (the equivalent of the mute/launch hold). Currently grid mode goes to the song level sound editor menu when you press Select, regardless of if a pad is held.

m-m-adams, 2024-03-16:
Oh you're right, I was thinking number of repeats

0beron, 2024-03-17:
In rows mode this menu is currently 'hold launch'+'press select'. 
Grid mode is different in that the pads (in green mode) are both the launch pad and the clip itself, so holding a pad and hitting select is more like holding a row in rows view. In blue mode it is exactly like holding a row - in green mode the grid has some additional behaviour to allow hold + tweaking gold knobs etc)

In rows mode, holding a row and pressing select just says 'CANT', and in grid view it either does the same (blue mode) or triggers the same sounds editor menu regardless of the held pad (green mode).

So I propose moving the launch type menu to be triggered by holding a row+press select, or in grid mode hold a pad and press select.
This puts it in a sensible, currently unused place where a config menu for all per-clip settings can go, and avoids the problem of having it on the status pads themselves which mute/unmute on press even if they open the menu. (Grid mode has configurable launch on press or launch on release in green mode).

This would implement it nicely for grid mode and make rows mode consistent.

0beron, 2024-03-17:
So I see three alternatives for where to put this menu in grid mode - 
Keep it the same as in rows mode, and use hold pad+select in green mode only. Comes with the same drawback of launching the clip if the grid is setup to launch on  press and not release.
Do what is proposed in the above PR, and use hold row + select in rows, and hold pad + select in grid blue mode (and probably green as well). This removes the existing convert to audio shortcut and changes behaviour from official firmware. The current function is of limited use (in my opinion, other's workflows may differ), and can be worked around by simply making a new audio clip and deleting the original. Were there future intentions to make the 'convert to audio' actually bounce a clip containing notes to audio?
Leave everything as it is, and add another separate mode to grid view that would allow reaching this menu.

0beron, 2024-03-17:
Opted for a new mode in grid view. Could nearly have got it into green mode, but the 'Allow Selection in Green Mode' setting means that sometimes green mode pads behave like the mute pads, and other times they behave like row pads instead.

sapphire-arches, 2024-04-03:
Should be resolved with the new yellow grid mode.

closed 2024-04-03 by sapphire-arches