SynthstromAudible/DelugeFirmware #820: Line in OSC automation issue as drawing previews strange pitch

rpc9943, 2023-12-13:
### Please describe the problem:

Drawing pitch automation on line in OSC changes the pitch as I am drawing a line in, to a pitch not matching the automation and releases as I let go.

### What is the expected behavior?

To it not changing the pitch when I draw automation. After I released the pads from this it doesn't even match the automation that it made as I enter in.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

Amadeus 1.0

### If possible provide the steps to reproduce the issue and upload additional media:

1. Set up line in OSC
2. Go to pitch automation
3. Enter in lines and hear the pitch changes as I'm entering the line even if cursor is not on the area I'm entering, at a pitch that does not match the automation I'm entering in. After release the drawing of lines of pads, the pitch returns to the correct automation

bobtwinkles, 2023-12-17:
Repro'd, it sounds like what's happening is we're immediately applying the tapped automation value which is then reset to the value under the playhead when the playhead crosses another automation node. This is probably what we want when the playhead is stopped but not when the transport is running.

m-m-adams, 2024-02-14:
Related - I noticed applying automation to local parameters while a note is held and playback is stopped stops the note

m-m-adams, 2024-03-03:
Fixed

closed 2024-03-03 by m-m-adams