SynthstromAudible/DelugeFirmware #823: Cannot rename instruments in Arranger view via NAME shortcut (7SEG only)

ok-reza, 2023-12-13:
### Please describe the problem:

It seems renaming is blocked on 7SEG in Arranger view. The keyboard shows up but no input is allowed.

### What is the expected behavior?

To be able to rename songs. This may be an oversight with OLED/7SEG behavior on the same firmware?

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

c1.0.0

### If possible provide the steps to reproduce the issue and upload additional media:

Open a previous song or start a new one.
Enter Arranger mode
Hold Audition Pad of a given instrument + hit NAME shortcut.
Keyboard appears, no input received.

m-m-adams, 2023-12-15:
On one hand this is a bug, on the other hand renaming instruments breaks the synth preset loading logic in both community and official. So while this is probably fixable we should fix loading first

lburygin, 2023-12-15:
Actually you can rename, try to just type your instrument name and press enter. It is just the letters you typing don't show up on the screen, that's a bug, I found it too recently

vcastellm, 2023-12-21:
Same here, renaming works but it's not displayed.

bobtwinkles, 2024-01-11:
We should fix the display issue for 1.1, though we probably can't fix the browser bugs before then....

closed 2024-04-16 by m-m-adams