SynthstromAudible/DelugeFirmware #828: Can only overwrite Songs, or give them a number higher than the latest Song while saving

sissyfus, 2023-12-15:
### Please describe the problem:

While saving you cannot give a song a lower number than the highest number already saved, you can only overwrite or make a new one with a higher number. So when I press save I can save as 209, but not 26 even when there is nothing saved on 26. If I save without loading a track first it will not show the dot indicating something is already saved. 

### What is the expected behavior?

On the official firmware all the lover numbers that are not in use show up without a dot and the ones saved show up with one. 
On the Community you can only see the ones with dots. 

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

deluge-c1_0_0.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Method 1:
Turn on the Deluge 
Load a track 
Hit Save 

Method 2:
Turn on the Deluge 
Hit Save

m-m-adams, 2023-12-15:
You should be able to type in whatever number you want and save. What firmware did you upgrade from? I believe this is unchanged from the current official RC, but if you're on an older 7seg firmware it might have behaved differently

m-m-adams, 2023-12-15:
Confirmed this matches the official RC. Workaround is to choose a name, which can be a number if you want

closed 2023-12-15 by m-m-adams

sissyfus, 2023-12-15:
I upgraded from the official 4.0 beta 4 which works well. I don't use the Querty so maybe it's a scroll issue, did you try that? 

I find it serious that the songs already there is lacking the dot when you just turn on and press save. Makes it easy to overwrite songs.