SynthstromAudible/DelugeFirmware #832: Creating an audio clip in arranger view then adjusting the clip causes a crash (E369 error)

baymud, 2023-12-15:
### Please describe the problem:

If you create an audio clip in arranger view then hold the audition pad of that lane and attempt to adjust the gold volume knob the Deluge crashes, displaying E369. 

This does not seem to happen if you create the audio clip in song view first (or just do the same adjustment in song view). 

### What is the expected behavior?

You should be able to adjust the volume of the audio clip lane in arranger view without a crash occurring. 

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

C.1.0.0

### If possible provide the steps to reproduce the issue and upload additional media:

- Create an audio clip in arranger view
- Hold the audition pad of that audio lane and adjust the volume knob
- Crash

[image]

bobtwinkles, 2023-12-16:
This is fixed by 073d023f27c94c84245db4b3195fdf34a93ea77f.

bobtwinkles, 2023-12-16:
Specifically we likely just need the drive-by fix (d2a4147448774a3f5d607bc03c72c33ab130e2e5)

bobtwinkles, 2024-01-06:
Fixed in 1.0.1 RC by https://github.com/SynthstromAudible/DelugeFirmware/issues/832

closed 2024-01-06 by bobtwinkles