SynthstromAudible/DelugeFirmware #839: MIDI learned controls can be corrupted due to invalid offsets in flash storage

bobtwinkles, 2023-12-16:
### Please describe the problem:

The offsets for flash are overlapping, so midi learned controls corrupt eachother

### What is the expected behavior?

MIDI learned controls should not corrupt eachother

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

0beron, 2023-12-16:
Fix to address this now in:
https://github.com/SynthstromAudible/DelugeFirmware/pull/843

bobtwinkles, 2023-12-16:
Fixed by #843.

closed 2023-12-16 by bobtwinkles