SynthstromAudible/DelugeFirmware #845: Waveform Loop Lock should have a visual indication that the loop is currently locked

bobtwinkles, 2023-12-16:
Rather than just relying on the popups, we should draw the pad grid differently somehow. Bikeshed proposal: Render the upper and lower halves of the loop indicators with different colors (cyan/magenta on the top/bottom of the start, and magenta/cyan on the top/bottom of the end maybe?)

ok-reza, 2024-01-15:
a cool idea visually, i think it would work well with only loop points but i think it would be confusing with the following consideration:

i totally think start/end point locking should be a thing also (length lock?). if your visual suggestion was implemented for both loop and length lock, i think it would get confusing to see half of, say, the column red at the start point. once you are zoomed in and can't see the context of both points, you may mistake it for the other point.

here are some rough mock ups of options:

1)[image]
your bikeshed proposal (maybe the bottom half colors should only be 3 rows to emphasize what the column represents (5 green and 3 red = start point)

2)[image]
when locked, the two columns share the same color (in this case, the color inbetween the two in an additive color space)

3)[image]
same concept as above but alternating

4)[image]
same concept but with only tops and bottoms changed (makes main color more distinct)

5)[image]
counter part's colors only on bottom to keep things clear if reading from top to bottom

6)[image]
same concept as above but with yellow/blue

7)[image]
alternating colors but only one color. yellow = locked. would need a special case when eg. green and cyan meet, yellow is ignored so green and cyan alternate.

after making the mock ups, i think 7 feels the best and is clearest.

M0R172, 2024-01-17:
I would prefer 6 or 2, the others look too busy to me, especially the alternating colors of 3 and 7.

The calm and clear look of 2 makes me wonder if the pairs should maybe *always* be the same color, and the do an additional marker as in 6 when they are locked together?

ok-reza, 2024-01-17:
good feedback!

the issue with matching pairs (eg. start and end, loop start/loop end) is that when zoomed in, you can't visually discern if the column represents the start or the end, we don't want users to think they're editing an end point but are actually editing the start point.

another option just occurred to me, something like 7 but no color (yellow) inbetween. just dotted cyan/magenta or dotted red/green

M0R172, 2024-01-17:
But now that I think of it - since selecting one when they are locked together is equivalent to "selecting both", the blinking would make a LOT of sense! 

OK, and then, how about additionaly indicating locking, to be visible when only one marker is on screen. by extending the "foot", i.e. the bottom of the marker, one or two pads towards the partner, like this:

[image]

PR #1614 (2024-04-01): Independent sample locking and improved lock rendering
Fix #845 
Fix #1043
commit b8d13778 (2024-04-01): Independent sample locking and improved lock rendering (#1614)

* sample_marker_editor: minor c++ modernization

* sample_marker_editor: use reference for multisample range

* make loop lock per-sample

Fixes #1043

* sample_marker_editor: flash both columns when loop is locked

Fixes #845

* maybe better marker rendering

* greyout pads outside range

* loop rendering changelog entry

* marker flash v3

* backout of commit 4a13bfd8cd3aec3a515ee8c4fb7b09b896b86899

* slightly nicer logic for marker rendering

* sample_marker_editor: fix marker names when sample is reversed

* remove extranious [... cut]
PR #1649 (2024-04-02): Independent sample locking and improved lock rendering
Fix #845 
Fix #1043
commit b1ca243c (2024-04-02): Independent sample locking and improved lock rendering (#1614)

* sample_marker_editor: minor c++ modernization

* sample_marker_editor: use reference for multisample range

* make loop lock per-sample

Fixes #1043

* sample_marker_editor: flash both columns when loop is locked

Fixes #845

* maybe better marker rendering

* greyout pads outside range

* loop rendering changelog entry

* marker flash v3

* backout of commit 4a13bfd8cd3aec3a515ee8c4fb7b09b896b86899

* slightly nicer logic for marker rendering

* sample_marker_editor: fix marker names when sample is reversed

* remove extranious [... cut]
closed 2024-04-01 by seangoodvibes