SynthstromAudible/DelugeFirmware #849: Add a default setting for slices generated by kit slice load

bobtwinkles, 2023-12-17:
Kits can be generated from samples by holding select when selecting the initial kit sample and scrolling to "SLICE". These slices currently generate kits where every sample is in "CUT" mode. However, multiple users have requested the ability to default to "ONCE." We should add a default to the system defaults menu which allows setting this mode to one of the existing slice modes (ONCE/CUT/LOOP/STRETCH).

trappar, 2023-12-17:
Also see: https://github.com/SynthstromAudible/DelugeFirmware/issues/510

joxihan, 2024-03-03:
i was looking at this issue but it seems like it's not cut mode by default but more :
firstDrum->sources[0].repeatMode = (lengthMSPerSlice < 2002) ? SampleRepeatMode::ONCE : SampleRepeatMode::CUT;
So it's a question of lengthperslice.
Does it mean that we want to change the logic to replace by whatever is set as default slice repeat mode from the UI ?

m-m-adams, 2024-03-03:
Yeah I think so - I think this logic is to try and guess whether it's a drum or a loop?

sichtbeton, 2024-03-22:
Yep, this logic assumes stuff that’s not apparent. Maybe could there could be a toggle to preserve it?

closed 2024-04-03 by m-m-adams