SynthstromAudible/DelugeFirmware #852: Kit randomization should not be limitted to just 25 samples

bobtwinkles, 2023-12-17:
I think we can do something more efficient if we get at the underlying fatfs nodes more directly, to remove the arbitrary limit on randomization.

M0R172, 2024-01-07:
First, thanks a lot for all your great Work here, and sorry to barge into the discussion as a deluge firmware noob!

This feature is close to my heart, but I have not gotten around to actually installing the build toolchain etc, so I just had some looks at the source code and the FAT specifications... Here are some observations:

1. "get at the underlying fatfs nodes more directly" - From what I understand of the FAT specs and the code in ff.c, this is currently done by sequentially reading 32 byte directory entries until it hits a zero entry. This does not seem like a particularly slow operation, and as far as I can see, there is no mechanism to get the number of files in a directory any faster on a FAT file system. 

2. This is especially true if we insist on checking every filename for being a valid audio file - this requires visiting every filename even if there actually *was* a faster way to get the total number. 

My conclusions on possible speedups: 

A. The code is using f_readdir_get_filepointer instead of f_readdir (see ff.c), but the "bonus" filepointer is never used. I doubt this makes a measurable speed difference, but it definitely is not required and very simple to change.

B. If the "isAudioFilename(staticFNO.fname)" check is slowing the operation down, one could think of dropping it - why would somebody put non-audio files in a samples directoy? Would failing (e.g. just not loading a new sample, i do not suggest crashing) be acceptable in this unlikely case?

C. If,as I am afraid, both do not relevantly help, and I am also correct in thinking that the actual io operation is as fast as it gets, one is IMHO left with buffering, leading to the design decision of how and when?

- For completeness I mention the possibility of having this info on the SD card somehow (e.g. a meta file in each sample directory), but keeping this up to date plus changing the sd content definition seems unattractive.

- One could scan the directory only on the first randomization and buffer it per kit row (or per directory but this seems more complex), (invalidating the cache on leaving the clip editor, or maybe on loading a sample from a different dir)? This makes some sense to me, as one will usually randomize several times before finding a good result: Only the first randomize would be slow, subsequent ones fast, and one would lose neither performance not memory when not using the feature.

- Doing it on first kit/sample load could also be an option, with the counter argument that one would pay speed and memory costs even if never using the feature, but the pro that also the first randomize qould be quick.

- One would still have to limit this somehow, as the possible worst case of scanning 65k? files is probably not acceptable from a memory and speed standpoint, no matter at which point in time the scan is done.

D. Maybe (I hope!) there is some much smarter and simpler way to do this, which I am completely overlooking... 

In any case, I would root for making the maximum randomization number user adjustable even if everything else stays as is, as different users (and sd card speeds) might have different opinions on what delay is acceptable for randomization.

Cheers!
M

M0R172, 2024-01-07:
Okay, so I finally tried it myself, installed the toolchain, changed the limit from 25 to 1000 and built the firmware:

I do not get any noticeable delay when randomizing from folders with several hundred drum samples - am I missing something? (BTW what would be the simplest way to time the execution of the function and display the resulton oled?)

I will try driving this to the limit with thousands of files in a folder, and more tracks running in parallel, but to me it definitely looks like this is fast enough as it is to increase the limit to far more than 25 without performance problems.

Running community FW 1.0 (only modified the randomization limit), release build, on an OLED Deluge with a 8GB class 10 ADATA SDHC card.

Cheers!
M

M0R172, 2024-01-07:
Ah, OK, with more tracks running in parallel I now get SD-card errors requiring a reboot...

bobtwinkles, 2024-01-09:
thanks for looking in to this! SD card errors indicate that this code is actually subtly broken and is somehow corrupting the file/cluster cache state, which isn't terribly surprising (as we've had reports related to crashing after using drum randomizers that never made much sense). If we can figure out why that is and fix it, maybe we can just remove the limit and leave it at that.

M0R172, 2024-01-12:
https://github.com/SynthstromAudible/DelugeFirmware/pull/955

M0R172, 2024-01-15:
Should this Issue be closed or the status updated?

sichtbeton, 2024-01-17:
This could be closed as it seems solved by #984 ?

sapphire-arches, 2024-01-18:
Yep, this was fixed by #955 (with a bit of followup work in #984).

closed 2024-01-18 by sapphire-arches