SynthstromAudible/DelugeFirmware #855: Audition and Mute Pads do not get rendered correctly when re-entering synths after saving a song

bobtwinkles, 2023-12-18:
### Please describe the problem:

Loading the song file from acidaliex on discord (https://discord.com/channels/608916579421257728/1138924035657449482/1186028879807258715) and entering the column 2 or 3 synths (in grid mode), the note row mute pads do not get rendered correctly.

### What is the expected behavior?

Note row mute/audition pads get rendered

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus, Nightly

### What is the firmware name:

fe31a4f9b112890b163157969097788eb81314ff

sichtbeton, 2024-01-07:
Had this but also a couple of times. When entering from grid mode, the sidebar goes blank. When you scroll up and down, it comes back.

m-m-adams, 2024-04-27:
I don't have a good solution but the problem happens [when the zoom is setup here](https://github.com/SynthstromAudible/DelugeFirmware/blob/3017c21195189b42787a8b8a80962a7d0b44be52/src/deluge/gui/views/timeline_view.cpp#L325) after the [fade](https://github.com/SynthstromAudible/DelugeFirmware/blob/3017c21195189b42787a8b8a80962a7d0b44be52/src/deluge/hid/led/pad_leds.cpp#L812)

At this point the sidebar has been rendered to the image store but not the image and gets wiped out. I believe that's intentional here so the sidebar probably just has to render again after the zoom?

sichtbeton, 2024-05-05:
I get this a lot. In every session. it's not a huge deal but kind of annoying.

closed 2024-05-15 by sapphire-arches