SynthstromAudible/DelugeFirmware #863: Stuck notes on MPE when Lower is configured to use all 16 channels and Upper is disabled

soymonitus, 2023-12-21:
### Please describe the problem:

I get stuck notes when learning a midi MPE keyboard (Lower) and then playing it, with channels setup to take the whole 16 channels (Upper zone disabled).

### What is the expected behavior?

I don't get stuck notes

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_0_0-nightly+2023_12_21-58b0d04

### If possible provide the steps to reproduce the issue and upload additional media:

Boot Deluge, connected to an MPE keyboard in the midi input and a midi monitor in the output
Set Upstream device 1 -> MPE -> IN -> Lower num of channels -> 15
Set Upstream device 1 -> MPE -> IN -> Upper num of channels -> 0
Set Upstream device 1 -> MPE -> OUT -> Lower num of channels -> 15
Set Upstream device 1 -> MPE -> OUT -> Upper num of channels -> 0
Create Midi clip and set output channel to MPE Lower
learn the keyboard with Learn + hold audition + press a key
Play any notes in the keyboard
See the midi monitor getting notes but never get the note offs, stuck notes

Repeat but setting MPE IN Lower NumChannels to 14 and MPE OUT Lower NumChannels to 14 and there won't be any stuck notes

m-m-adams, 2023-12-21:
1. Deluge is assigning notes to channel 16 more often than it should
2. Notes assigned to channel 16 don't get a note off
3. This happens without an MPE controller, only MPE output is required to repro the bug

m-m-adams, 2023-12-21:
The problem is note offs not getting sent on channel 16, that compounds with how MPE channel allocation works to cause the other issue

PR #864 (2023-12-23): MPE fix - allow note offs on channel 16
Fix #863
closed 2023-12-23 by seangoodvibes