SynthstromAudible/DelugeFirmware #866: Default Bend Range doesn't seem to apply to Poly / Finger / MPE

roge-rm, 2023-12-22:
### Please describe the problem:

I got myself an Exquis and it doesn't play nice with the default MPE bend range of 48. 

There doesn't appear to be an option to change the default bend range for MPE devices, though under Settings->Defaults there is a Bend Range option... However it only appears to affect the "Normal" bend range and not the "Poly/Finger/MPE" bend range.

### What is the expected behavior?

I think I epected one default bend range setting to rule them all. I would also be happy with a separate setting specifically for Poly/Finger/MPE

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus, Nightly

### What is the firmware name:

1.0.0-NIGHTLY-58B0D04

### If possible provide the steps to reproduce the issue and upload additional media:

[image]

step 1: buy Exquis
step 2: look how cool they look together!

m-m-adams, 2023-12-22:
Changed title as there should be two default ranges, the existing one (for pitch wheels) and the MPE one (for MPE/per finger slides). They need to be adjusted seperately as MPE bends can require much larger values than wheels

This should really be a per device setting as well, as it can vary between controllers. For example the linnstrument typically uses 48 which would be awful on the osmose or exquis

roge-rm, 2023-12-22:
Yeah, where you suggested to me it should be (under MIDI->Devices) makes a lot of sense for that reason. Thanks :)

soymonitus, 2023-12-25:
Maybe I would like to work on this as I have an Osmose and don't like how it plays with default 48 value.
Could anyone tell why the default value for the wheel is 2 and for fingers is 48? are both measured on semitones? or do they have different units? the main bend range it is clear to me that 2 is number of semitones, but what is 48 then? because when I move my finger on the osmose I would say Deluge synths slide up or down 1 or 2 semitones at most

m-m-adams, 2023-12-26:
They're both in semitones - MPE has a much larger range because on devices like the Continuum, Linnstrument or roli boards you can slide multiple octaves, which is sent as a pitch bend 

Deluge responds to the MPE bend range configuration method if there is a way to send it from the osmose, that's probably your quickest option

soymonitus, 2023-12-26:
After testing to hardcode finger range to 24 i found that it is the same as half semitone, the same setting i have on the osmose

There is already a setting in synth presets for finger range!!

m-m-adams, 2023-12-26:
Can you look in a MIDI monitor at what the osmose is actually sending? If what you're describing is accurate then it is not following the spec and will be tough to support

soymonitus, 2023-12-28:
Osmose sends pitch bend finger center value as 8192, then pushing the key all the way possible to the left it reaches 8022, and to the right 8363. If I set local off on the Osmose, and play it through the Deluge, with a learned Midi Lower Zone track, all i can get is a 2 semitones pitch bend per key. It doesn't matter how I set the bend range on the Osmose (1/2semitone for example), it is always moving in +-2 semitones range. I don't know if it is a fault of the Osmose or of Deluge.
Would it make sense to also include a finger bend range for midi mpe tracks?

soymonitus, 2023-12-28:
If, on the other hand, I record something in the Deluge in MPE mode, and then change the Midi track to a synth track, then I have control in the synth settings to change the finger bend range and can set it from default 48 to 24 and then I can get the desired 1/2 semitone I want...

m-m-adams, 2024-01-06:
> Osmose sends pitch bend finger center value as 8192, then pushing the key all the way possible to the left it reaches 8022, and to the right 8363. If I set local off on the Osmose, and play it through the Deluge, with a learned Midi Lower Zone track, all i can get is a 2 semitones pitch bend per key. It doesn't matter how I set the bend range on the Osmose (1/2semitone for example), it is always moving in +-2 semitones range. I don't know if it is a fault of the Osmose or of Deluge. Would it make sense to also include a finger bend range for midi mpe tracks? I am guessing no, because what you record in the MPE deluge track is what you output later, no trick there. I am asking for help in the Osmose facebook group, maybe I am missing something

This sounds like an osmose limitation - for midi tracks Deluge sends out the pitch bend exactly as received

m-m-adams, 2024-01-08:
2 step fix:
1. Incorporate per-finger bend range into the pitch automation data rather than storing it separately. Rather than storing the pitch bend and separately a multiplier from 1-48, store the pitch bend * per finger bend range. This will keep performances captured as played and allow changing settings on synths and switching between MPE controllers without breaking previous recordings.
2. Move per-finger bend range from an instrument attribute to an input attribute

To make the transition some extra code will be required to combine the two attributes in existing song files on load

N.B. - breaks compatibility with official at this point, but I think it's necessary as the current setup is severely limiting. For example you can't currently record from linnstrument (where +-48 semitones is normal) and then add some extra notes from an exquis (where +-1 is normal)

m-m-adams, 2024-01-08:
Actually we could set per finger bend to 1 in the song to keep compatibility with official afterwards

closed 2024-12-28 by m-m-adams