SynthstromAudible/DelugeFirmware #875: Arpeggiator should not be an option of the Kit/Affect Entire FX menu

soymonitus, 2023-12-23:
### Please describe the problem:

There is an option to change Arp settings from the new added Affect-Entire-Kit menu. When changing mode Deluge freezes. Besides, why arp settings for affect entire? Shouldn't the menu be completely removed?

### What is the expected behavior?

There is no arpegiator menu in Affect Entire Kit menu

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

soymonitus, 2023-12-23:
@seangoodvibes You developed this menu I think so asking you if there is a reason to have arp settings available

seangoodvibes, 2023-12-23:
> @seangoodvibes You developed this menu I think so asking you if there is a reason to have arp settings available

I added it because Arp Gate is available as a Kit Affect Entire param, but I'm not sure if it does anything. If it doesn't it can be removed.

Probably same for stutter rate.

soymonitus, 2023-12-25:
Why would arp gate affect a bunch of kit rows which could even not have arp ON? It doesn't make much sense to me, even if it works somehow because the code paths allow it, it is not intuitive at all and should be removed

closed 2023-12-30 by seangoodvibes