SynthstromAudible/DelugeFirmware #914: Adjusting Volume of Audio Track created in Arranger view crashes Deluge

OaklandGhosts, 2024-01-05:
### Please describe the problem:

Happens 100% of the time for me, even in blank session.
Step 1: Create a midi track to control external synth. 
Step 2: Create an audio track for monitoring in Arranger View and set line in to Stereo (Monitoring)
Step 3: Adjust Volume by holding down audition pad for audio track in arranger view and twisting top gold knob for volume.

Deluge Crashes - Error E369

If you first create the audio track in song view and then go to arranger view and try to adjust volume it works as expected.

I'm creating the audio track in arranger view because I ONLY want it for monitoring and not recording. But doing so crashes when I adjust volume. This did not happen previously with Factory 4.0+ firmware

### What is the expected behavior?

Expect to be able to adjust volume of an audio track that is set to monitoring in arranger view.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### If possible provide the steps to reproduce the issue and upload additional media:

Step 1: Create a midi track to control external synth. 
Step 2: Create an audio track for monitoring in Arranger View and set line in to Stereo (Monitoring)
Step 3: Adjust Volume by holding down audition pad for audio track in arranger view and twisting top gold knob for volume.

Deluge Crashes - Error E369

m-m-adams, 2024-01-05:
This is already fixed if you download a nightly build, we'll put up an updated release with the fixes soon

closed 2024-01-05 by m-m-adams

sichtbeton, 2024-01-05:
Fixed in nightly