SynthstromAudible/DelugeFirmware #941: Voice Dropping / Sample dropping

Evilpilli, 2024-01-09:
### Please describe the problem:

I've got a couple of songs that after upgrading to amadeus some voices and samples are dropped. The first does not use any multisamples, while the second one does. 

### What is the expected behavior?

On earlier versions of the community firmware, these songs would playback completely fine, without any dropped voices or samples, even with the grainfx engaged. 

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

Amadeus 1.0

### If possible provide the steps to reproduce the issue and upload additional media:

Files were too big to upload here (might be part of the issue?) 
So I made a google drive link for the song examples. 
https://drive.google.com/file/d/1JwuULpZ-l3bJVDdWSSXilbMw9zhxSGbN/view?usp=drive_link

songsta, 2024-01-13:
This is now the 3rd issue relating to this. 

1. https://github.com/SynthstromAudible/DelugeFirmware/issues/816
2. https://github.com/SynthstromAudible/DelugeFirmware/issues/804

Perhaps we should merge them?

This for me is the one issue that is stopping me from adopting the new firmware.

chewabledrapery, 2024-01-19:
Hi,

As requested by reza in the Deluge facebook page, here is a project that is dropping voices, in this project with all clips the kick drum is being dropped occasionally, swapping between some samples clips exacerbates the issue.

Hope this helps the team to track down the issue, all the best and thanks for all the amazing work!

[Example1 2.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/13990726/Example1.2.zip)

closed 2024-01-19 by m-m-adams

Evilpilli, 2024-01-19:
Not feeling forgotten, it seems like an esoteric issue that's hard to pin down. Thanks for taking the time!