SynthstromAudible/DelugeFirmware #942: Saving tracks causes playback/modification issues

elijackson97, 2024-01-09:
### Please describe the problem:

Hi, 

I have been using the community firmware and have noticed a persistent issue that is driving me a bit mad. When I am working on a track, I create a few clips, create a new synth sound, and add drums etc. I'll then go to save the track under a name like "p trackname" 

The problem arises once I load the track back in. I get the E044 error on just about every track I open under this new firmware. Then, for some reason, certain clips are unable to be played along with the other clips that I was previously playing together. (ex. I had 2 synth clips, one an arp and one a ring mod synth, and a drum track with a sample kick and a synth kick from my synth library, when I save and close, my arp synth can not be played along with my ring mod synth at the same time.) I have also lost significant modifications to my ring mod synth sounds and the track sounds drastically different. 

Another issue I have noticed is that when making a track in grid mode and after saving the track, the grid no longer is in the same format as before. I am unable to play certain tracks that were previously playing, and unable to launch the different clips as when they were being made in grid mode. 

I am loving all the new features but these bugs are frustrating. If anyone has any advice on how to avoid this, it would be greatly appreciated. 

Thank you

### What is the expected behavior?

Saving created tracks load in the same state as they were before saving. Grid mode retains format after saving. 

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

Not positive, file is 1288 KB file called deluge-c1_0_0

bobtwinkles, 2024-01-09:
Hi, thanks for the bug report!

If you're getting an Exxx error, the pad grid should light up with some patterns. If you can provide a picture of those patterns (either here or in the #nightly-testing channel on Discord) that would help really help us narrow down the problem. 

For the grid mode problems, can you share the "collect all" song file? One common issue is if you have multiple rows of the same instrument in the same section (you can check this via Rows mode) Grid will only be able to display one of them. This is a UI limitation we'd like to remove eventually, though there's no timeline as of yet.

For future reference, you can find the exact firmware version by pressing shift+select to get to the main settings menu, then scrolling to the end of the list. Assuming you downloaded it from the 1.0.0 release tag that should be enough information. Also, maybe try out the [new 1.0.1 release](https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/release_1_0_1). We fixed a few crash bugs, so maybe things will work better for you there.

m-m-adams, 2024-01-11:
Looked at the song file, the bug is that synths created with shift+synth aren't getting unique names

m-m-adams, 2024-01-11:
The rest is intended behaviour for how to handle loading 'impossible' song files

closed 2024-03-03 by m-m-adams