SynthstromAudible/DelugeFirmware #947: Grid view doesn't get updated as song is being played from arrangement view

wargoth, 2024-01-10:
### Please describe the problem:

When song is played from arrangement view and I switch to grid view it displays currently playing clips as expected. However as the song advances and new clips start and stop playing the grid view doesn't get updated. Workaround is to use scroll encoders which force the view to update.

### What is the expected behavior?

When song is played from arrangement view and I switch to grid view it displays currently playing clips, and should keep the view up-to-date, just like the row view.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create a couple of clips
2. Arrange them in song arrangement view so they start/stop mimicking song progression
3. Start song in arrangement view
4. Switch to grid view. Note it doesn't get updated as clips start/stop

bobtwinkles, 2024-01-10:
This should be a reasonably straightforward fix, once someone has time to look at it. There's probably a cache somewhere in either the pad rendering or the play state it self of the session view that isn't getting invalidated appropriately.

bobtwinkles, 2024-01-14:
Dupe of #789

closed 2024-01-14 by bobtwinkles