SynthstromAudible/DelugeFirmware #978: MIDI recording from arranger doesn't get note length

colinmeyer, 2024-01-15:
### Please describe the problem:

1. New song
2. Midi clip
3. Arranger mode
4. Make sure midi track is armed for recording
5. Record, then play, press some buttons on external controller, to get some midi notes recorded to a newly minted fresh white clip (i.e. let the record function create the white clip, don't create it ahead)
6. Stop, enter newly created white clip. Observe that notes (at least after the first note) have the minimum lenth, instead of the proper length that the button was pressed

### What is the expected behavior?

All notes should record their proper length, when being written into a new white midi clip

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

a5a6fd056df41ef7f713e5642e5b338ae3b0f5dc

### If possible provide the steps to reproduce the issue and upload additional media:

see above

Radius9999, 2024-02-01:
I discovered this does not happen if quantization is set to none.

m-m-adams, 2024-02-26:
This is a quantization bug - the menu allowed setting impossible values which caused issues in some scenarios

closed 2024-02-27 by seangoodvibes

Radius9999, 2024-02-29:
#1348 did not fix the issue [Deluge Beta Release 20240229]

PR #1361 (2024-02-27): limit quantization to values which work
In official the list is 10 items - the triplets/dotted support extended that, but in the case of quantizaiton they don't work and cause notes to fail to record

Fix #978