SynthstromAudible/DelugeFirmware #1002: Can not load songs or samples after trying to write to read-only SD card

M0R172, 2024-01-16:
### Please describe the problem:

See title. Discovered this while working on #1001.

### What is the expected behavior?

You should be able to load songs, kits, and samples, even if the SD card is in read-only mode.

However, after the first SD write error, something seems to be corrupted, so loading also does not work anymore.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### If possible provide the steps to reproduce the issue and upload additional media:

- Make SD card write-protected (small switch on side of card) 
- put into deluge
- You can load songs as usual
- Now try saving a song, you will, correctly, get a sd write error
- *Now* try loading a song, or switching kits, you will get sd card error or no files found

m-m-adams, 2024-01-17:
While there should be a better error (probably a check on boot?), this case will have weird errors regardless. The SD card is used to cache samples and audio recording (even if the song is not yet saved), and without that ability Deluge will have memory issues

M0R172, 2024-01-17:
> While there should be a better error (probably a check on boot?), this case will have weird errors regardless. The SD card is used to cache samples and audio recording (even if the song is not yet saved), and without that ability Deluge will have memory issues

OK, then the user should really be warned strongly in this case, and not just notified as in #1001

Would be easy enough check on boot, and just stop with a message if card is in RO. If you think it makes sense, I'll make a PR for that later today. 

My only issue is that you could still remove the SD while Deluge is running, make it RO and put it back in, but I guess if you do that on purpose you deserve to suffer... ;-)

M0R172, 2024-01-17:
Another point: The Deluge will happily boot and operate without a SD card put in. If you try saving something, you get an error, but can continue unhindered otherwise.

I always thought this was pretty cool, but if

> The SD card is used to cache samples and audio recording (even if the song is not yet saved), and without that ability Deluge will have memory issues

This means the user is operating on borrowed time and things could just start failing, right? So shouldn't we also prevent booting, or at least give a stern warning when not having any SD card?

m-m-adams, 2024-01-18:
Ya without an SD card you'll only be able to record 4-5 minutes of audio before you have issues

sapphire-arches, 2024-02-03:
Fixed by #1164

closed 2024-02-03 by sapphire-arches