SynthstromAudible/DelugeFirmware #1006: Noticeable lag when choosing synth presets in song with minimal clips

hsensh, 2024-01-17:
### Please describe the problem:

I have a Deluge 7SEG device with Bootloader version 006 running the 1.0.1c firmware and in a relatively simple song, it's lagging when choosing synth presets

### What is the expected behavior?

The Deluge shouldn't lag when choosing synth presets considering the lack of any complexity or anything memory consuming in the song.

### Is there a relevant Pull request?

Not that I can find

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

C 1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

I was tinkering with a song and what I remember doing is:

- Added some pan automation to a track using the community firmware automation view then deleted it
- Used the community firmware drum velocity view to record some steps

I then went to the song view and created a new synth clip and was scrolling through the presets and there was a noticeable lag when switching between them, like the presets were taking time to load. If I go a bit too fast it even skips a couple and doesn't play right away.

Some things to note:

- Everything in the community settings was turned on except SYSX, NORN, and GRFX. AUTO sub-settings were left at their defaults.
- This only happens while the song is playing, when it's paused everything goes back to normal.
- It's not a huge inconvenience and it's manageable, but I was afraid it might be some sort of memory leak because I have way more complex tracks with significant automation among other things and the Deluge never lagged or struggled, so this happening in a very simple 6-track song was a bit alarming.

Here is the XML file for the song:

[SONG014.XML.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/13957869/SONG014.XML.zip)

Let me know if you need anything else from the SD Card.

Thank you for all your hard work!

sapphire-arches, 2024-01-17:
This might be a dup of #787. We applied a big hammer to fix some gremlins in the browser code, and it reduced browsing performance to a noticeable degree in some circumstances, unfortunately.

hsensh, 2024-01-17:
Let me know if there's any way I can help. I know my way around C++ and I can manage to understand the issue if I dig in. Is there anything I should be looking for?

sapphire-arches, 2024-01-17:
You'd probably want to take a look at `browser.cpp` and `slot_browser.cpp` -- the slot browser supports old (pre 3.0 IIRC) SD card layouts while the Browser is the primary interface for browsing SD contents. You'll want to look at #715, #718, and the fix for those issues, #719, to understand the context. If you're on Discord we hang out in #dev-chat on the Synthstrom community discord and you can usually get quick responses there. If you do decide to take on the browser bugs we'd be eternally grateful as a couple people have banged their heads against it and not made a huge amount of progress. Happy hunting, bring a bright torch :smile:

hsensh, 2024-01-18:
Thanks for the insight, I'll take a look at them when I have some time on my hands. I'll join you guys on discord since it'll take me way longer to figure all this code out myself. Looking forward!

sapphire-arches, 2024-01-27:
Closing in favor of #787 since I think this is the same bug.

closed 2024-01-27 by sapphire-arches