SynthstromAudible/DelugeFirmware #1008: Move isParamPan/isParamPitch/isParamQuantizedStutter/isParamStutter out of View Class

seangoodvibes, 2024-01-17:
### Please describe the problem:

These functions should not be part of the view class.

### What is the expected behavior?

They should belong to another more appropriate class as freestanding functions. E.g. param class?

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

m-m-adams, 2024-01-19:
I think the most appropriate spot would be to make them free standing functions in the param namespace

sapphire-arches, 2024-01-19:
This was done as part of #1011, specifically https://github.com/SynthstromAudible/DelugeFirmware/pull/1011/commits/b193a731de9c977e8a70eeb6edece20310052b94

closed 2024-01-19 by sapphire-arches