SynthstromAudible/DelugeFirmware #1030: Keyboard in file browser not showing when scrolling with horizontal encode

sichtbeton, 2024-01-21:
### Please describe the problem:

When in the file browser you scroll with the horizontal encoder, the name bar is shifting. When you then type with the pads, it types but the keyboard is not present.

### What is the expected behavior?

The keyboard should come up.

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.1.0-DEV-85997821

sichtbeton, 2024-01-21:
https://github.com/SynthstromAudible/DelugeFirmware/assets/62035517/86356844-1216-4b66-98ae-61225223429e

sapphire-arches, 2024-01-27:
This bug is actually quite old, potentially even going back to the initial OSS release.

sichtbeton, 2024-01-29:
> This bug is actually quite old, potentially even going back to the initial OSS release.

And obviously didnt bother anyone since :-)

closed 2024-02-01 by stellar-aria