SynthstromAudible/DelugeFirmware #1046: Quantizing a row or all rows should not change value for the first turn.

ok-reza, 2024-01-23:
### Please describe the problem:

Quantizing a row or all rows by one notch will increase/decrease the value each time.

### What is the expected behavior?

To maintain consistency and user-friendliness, the first turn should always display the current value like iteration dependence does. This will make it so a user can easily check the current value, then decide if it needs to be adjusted.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0 - 8FEA0F5

### If possible provide the steps to reproduce the issue and upload additional media:

1) Enter a kit/synth/midi clip.
2) Press and hold an audition pad and turn TEMPO (or press and turn TEMPO for all rows)
3) Note that quantize or humanize is now at 10% instead of first displaying 0%.

Probably also related and would get fixed if the above is addressed:
1) Set quantize to 100%
2) Try to increase quantize again (clockwise)
3) Since quantize is already set to 100%, the value doesn't update. However, that means the user will not see the display confirm it's at 100%. This causes confusion over whether the feature is working or not.

m-m-adams, 2024-01-23:
Not fixable without overhauling the humanize/quantize system - the humanize/quantize value only exists while adjusting, after adjusting it's written to the underlying notes

closed 2024-01-23 by m-m-adams

m-m-adams, 2024-01-23:
That issue is tracked in #401