SynthstromAudible/DelugeFirmware #1051: Converting empty Audio Clip to Kit crashes, in some scenarios it will reverse play direction of the newly created Kit.

ok-reza, 2024-01-24:
### Please describe the problem:

Converting a blank Audio Clip to Kit usually will generate a E379 error. There is a chance to convert this successfully, and in some cases it will reverse the play direction of the Kit.

### What is the expected behavior?

Converting Audio Clip to Kit should properly load the first available kit. Play direction should not change when converting.

### Is there a relevant Pull request?

Possibly related to #760

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0 - 8FEA0F5

### If possible provide the steps to reproduce the issue and upload additional media:

For E379 error:
1. Turn on Deluge, enter SONG view
2. Press and hold row pad and convert to Audio Clip by pressing SELECT
3. Press and hold row pad of newly converted Audio Clip and press KIT
4. Crash should occur.

For Play Direction switch:
1. Turn on Deluge, enter SONG view
2. Create a new row and convert it to an Audio Clip
3. Press and hold row pad of newly converted Audio Clip and press KIT
4. If a crash doesn't occur, the play direction can reverse. Sometimes takes a couple tries.
5.  Additionally, the kit made doesn't properly load all the samples: in my case it loads 4 samples of the 808 preset, the first two and last two. (see https://github.com/SynthstromAudible/DelugeFirmware/issues/760)

m-m-adams, 2024-01-24:
I'm inclined to just block this - is there a reason to do it instead of deleting the audio clip and making a new clip?

ok-reza, 2024-01-24:
i think the use case is exactly that, to convert an empty audio clip to another clip type quicker than doing it manually. the use case is probably low specifically into KITS, but we are able to convert any empty clip to any other clip type currently so i think it's a nice to have. i don't think it would be a problem to have it blocked simply for the 1.1 release for stability purposes but addressed in the future.

the only other use case i can think of is if someone has gone ahead and made a template project with a variety of blank clips, and decides they don't need an audio clip and quickly wants to convert it.

thinking into the future, freezing/unfreezing tracks could work with selecting a sequenced clip's pad row and then pressing SELECT (like you would to create an audio clip, but in this case it freezes the eg. synth clip into an audio clip) and vice versa, but i imagine there may be some UX gremlins that pop up with this.

in any case, this is stable in official firmware. also it seems to be stable from blank audio clip -> blank synth clip (no crashing) but i was able to have play direction reverse once doing this.

m-m-adams, 2024-01-24:
Official has a variety of other bugs around converting clips to kits, so it's possible this one just presents itself differently there

seangoodvibes, 2024-03-21:
Closed as should be fixed with PR #1532

closed 2024-03-21 by seangoodvibes