SynthstromAudible/DelugeFirmware #1060: While in Grid View (Blue Mode), holding a colored section's launch pad and then entering a clip will display a pop up until another pop up occurs.

ok-reza, 2024-01-24:
### Please describe the problem:

When holding a colored section's launch pad in Grid View (while in blue mode) will then display the launch status "Repeats: Infinite". While this is displayed and you enter a particular clip, the pop up will display indefinitely until another pop up replaces it (eg. entering a note "C#3" will clear the previous pop up.

This behavior is OLED only.

### What is the expected behavior?

"Repeats: Infinite" pop up should disappear when entering a clip.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0 - 1AD7933 (1/24)

sapphire-arches, 2024-01-27:
The fix in #1070 probably should have picked this up as well, the reason why it didn't probably needs to be investigated.

m-m-adams, 2024-03-03:
That fix is for entering sound editor, I don't think a clip is a sound editor until you open a shortcut

closed 2024-03-20 by m-m-adams