SynthstromAudible/DelugeFirmware #1062: 7SEG lights flicker oddly when pressing certain buttons

ok-reza, 2024-01-24:
### Please describe the problem:

7SEG only
Pressing certain pads (generally ones that don't emit text on the display) seems to illuminate random segments on the display.

### What is the expected behavior?

Random segments shouldn't flicker when pressing certain pads.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly, other

### What is the firmware name:

1.0.1, 1.1.0 - 1AD7933 (1/24)

### If possible provide the steps to reproduce the issue and upload additional media:

In this case, SONG (Rows View), press audition pad repeatedly
or
SONG (Grid View), press audition pads repeatedly (any pad on the column where green and blue mode are)
https://github.com/SynthstromAudible/DelugeFirmware/assets/136033640/d921480a-d7db-4b34-957b-f2fd03b8af76

sapphire-arches, 2024-01-26:
This is no-repro on the oled 7seg emulation, so it's probably something going wrong in the raw PIC command stream. Since it also happens on 1.0.1 I don't have any idea what caused this, and bisecting might be hard.

ok-reza, 2024-02-01:
update: i found out a possible culprit, if i make the flicker occur super fast it seems like the display is trying to display "SONG", maybe it's related to the 'SONG' and 'ARRANGER' text that now shows up on OLED for song and arranger view.

stellar-aria, 2024-02-15:
For reference this looks like a bad interaction in SessionView::redrawNumericDisplay() and renderViewDisplay. Also the strings being referenced seem to have no 7SEG version so they're trying to scroll

PR #1460 (2024-03-10): session: do not show info names on 7seg
Maybe fixes #1062
commit faed11c0 (2024-03-10): session: do not show info names on 7seg (#1460)

Maybe fixes #1062
PR #1465 (2024-03-10): session: do not show info names on 7seg
Maybe fixes #1062
commit e2c69f1a (2024-03-10): session: do not show info names on 7seg (#1460)

Maybe fixes #1062
closed 2024-03-10 by seangoodvibes