SynthstromAudible/DelugeFirmware #1069: new default for naming is broken

RadioChad, 2024-01-25:
### Please describe the problem:

Dault naming used to be simple and straight forward. Saving a song was like having a "states" function : by quickly (1 action) saving the various moments of a song creation, we can also easily go back to the previous "states" while keeping eveything clear to our eyes when using a 7-SEG sceen. it is a major feature of the vanilla Deluge.
The Community Update entirely destroyed that : When the new default naming works, the nomenclature breacks its usability on 7-SEQ displays if the number is bigger than 99. But even worse, soften it bugs (i think it is with big numbers, to verify) and the naming makes the whole thing unusable : project 185 becomes for some reason 185_185, then 185_186 and so on. The only workaround is to manually assign a name, which obviously put an end to the major feature I described above.

### What is the expected behavior?

Please read above

### Is there a relevant Pull request?

I don't know

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

c1

### If possible provide the steps to reproduce the issue and upload additional media:

Just save a song created on the fly. Try with 1 digit, 2 digit and 3 digit numbers to see what happens.

m-m-adams, 2024-01-25:
Once you have assigned a name the default incrementing works fine. Just name your song once

RadioChad, 2024-01-25:
The principle is if you have to do more actions to have the same result as how it was before, it is a downgrade. Maybe you could propose a vote to the community (since it is a community build) to know if the majority want some of the major features being downgraded or not ?

m-m-adams, 2024-01-25:
I am offering you a workaround. Anyone is of course welcome to contribute code and fix it

RadioChad, 2024-01-25:
> Anyone is of course welcome to contribute code and fix it
Yes that's why I posted that here :)

Your answer triggered me because on top of the obvious downgrade is the presence of a legit bug. The fact that it eventually add a very high number instead of _2, _3 etc, is not normal. Numbering should never start with something like 185.

sapphire-arches, 2024-01-25:
There is probably 2 separate bugs here:
 1. we're not properly detecting an old-style (slot browser) SD card layout. Need a listing of the file and directory structure on the card to confirm this. The card is also likely to never detect as an old-style SD card until the newer named (`186_186.xml` etc.) files are renamed, this can be confirmed by switching back to official
 2. In this partial slot browser state, we apparently start the song iteration number with the slot index instead of 0.