SynthstromAudible/DelugeFirmware #1073: Unable to disable MOD FX on audio channel

manysounds, 2024-01-25:
### Please describe the problem:

After enabling a modulation effect for an audio input track, the option to disable disappears.

### What is the expected behavior?

The MOD FX page would show “disabled” as an option.

### Is there a relevant Pull request?

Unknown

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

C1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

Power on
Create audio track with input of any type.
Enter clip mode for the track.
Enable MOD FX.
Move the select knob and disabled disappears.

ok-reza, 2024-01-26:
i brought this up during official firmware beta testing. just letting it be known that this is old, but still undesirable behavior.

additionally, audio clip's can access any mod fx parameter even if the current mod fx type selected doesn't use that particular mod fx parameter. (for other clips, irrelevant mod fx parameters are blocked from being accessed depending on mod fx type that is selected [eg. chorus enabled, can't enter feedback parameter, but you can enter feedback in audio clips (has no effect)])

sapphire-arches, 2024-01-26:
I believe this was fixed by #908, so it should be in 1.1 (or it's already fixed in the nightly if you're brave and have good SD card backups).

closed 2024-01-26 by sapphire-arches

manysounds, 2024-01-26:
Thank you for being ahead of the curve on this :)