SynthstromAudible/DelugeFirmware #1084: OLED is not correctly reset on boot

sapphire-arches, 2024-01-27:
Occasionally after flashing (either via sysex or via the SD bootloader) the OLED column pointer ends up somewhere offset and the whole display is aligned. In principle the `0x22` and `0x20` commands should reset the column pointer (according to to SSD1306 manuals), but apparently the display controller on our display is not a complete implementation.

m-m-adams, 2024-03-03:
From some testing adding a 100ms delay as soon as the timers are initialized works. I think debugging has the same issue as with yesterdays nightly where the dma hasn't completed. Since the dma from the previous run isn't stopped by the load we can wait for it to finish before resetting the dma controllers later in the startup routine

sapphire-arches, 2024-03-04:
We might be able to detect that the DMA engine is still active by checking the `EN` bit of the DMA engine - according to figure 9.3 of R01UH0437EJ0600 Rev.6.00 (RZ/A1L TRM) the DMA engine should reset that bit when the transfer completes. Adding a bit of code to `start.S` to wait for the DMA channel used by the bootloader and the main firmware (DMAC 4) to go idle.

https://github.com/SynthstromAudible/DelugeBootloader/blob/6ce8601eef539bcc3ad91432a1b8779a19aaaae5/src/drivers/oled/oled_low_level_all_cpus.h#L34

m-m-adams, 2024-08-08:
Technically still a problem but we don't really have a fix and the current state is fine

closed 2024-08-08 by m-m-adams