SynthstromAudible/DelugeFirmware #1153: Bug - resetting Deluge to factory default settings by holding Select and powering on doesn't work

seangoodvibes, 2024-02-01:
### Please describe the problem:

Can't reset the Deluge to factory defaults by holding down the Select encoder and powering on the Deluge

### What is the expected behavior?

You should be able to factory reset the Deluge by holding down Select and powering on the Deluge.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

m-m-adams, 2024-02-04:
https://github.com/m-m-adams/DelugeFirmware/blob/5c6e3812687db63d8d22cc360741fa9a2a85f07c/src/deluge/deluge.cpp#L631 

Problem appears to be here - I don't think it actually gets hit

M0R172, 2024-02-04:
Confirmed by print debugging: The case gets hit, it just never `looksOk`... 

Without the `looksOk` check, any combination of presses that includes the select button triggers factory reset...

closed 2024-02-05 by m-m-adams