SynthstromAudible/DelugeFirmware #1161: Weird issues with digital delay

x3003, 2024-02-03:
### Please describe the problem:

1) The sound becomes louder than it should compared to the analog style one.
2) And the sound won't stop when press play a second time.

### What is the expected behavior?

The sound should not get louder.
The sound should stop when press play a second time.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

Deluge Nightly Release 20240202

### If possible provide the steps to reproduce the issue and upload additional media:

1) Take a stock sound, like 000 rich saw bass.
2) Fill 1 row completely.
3) Set delay at 25
4) Press play, the sound should play normally as it's using the analog delay.
5) Switch to digital delay. Now the sound becomes [much] louder.
6) Press play again to stop the sequencer. The sound still is playing continuously.

If the delay is set to 24 or lower the sound stops.

sapphire-arches, 2024-02-03:
This is expected behavior, I think -- at the very least it's how official works. Digital delay with "amount" at 25 is a lossless feedback loop, so it will continue to make sound until the gain is set to something less than 1. The analog delay has some filters in the feedback path which remove energy from the signal so it gradually dies out even at amount 25, but you can get it to go in to positive feedback by setting the amount to 26 or 27. We do want to implement an emergency-stop-all-sound shortcut but it's not implemented yet.

m-m-adams, 2024-02-04:
Yeah - this is intended behaviour. Closing the bug, but we are looking into ways to make this more intuitive - most likely having the self oscillation be at 45 or so instead of 25 on the knob. Sadly it's a little complicated in practice as we have to make the change while keeping compatibility with past presets.

FWIW there are people who make use of the self oscillation musically! It can be good to allow it to build up for effect and then drop the amount back down to stop

closed 2024-02-04 by m-m-adams