SynthstromAudible/DelugeFirmware #1162: MIDI-FOLLOW receives on all channels via USB MIDI / Learn

schtaf, 2024-02-03:
### Please describe the problem:

With both current community release and 1.1.0-Nightly-33B625E. My use case is arturia keystep controlling Deluge via midi over usb. When I learn to the desired channel from the channel submenu, the deluge will work as expected in Midi follow mode but will respond to any midi channel from that device. This breaks the utility of midi follow mode. Open to help troubleshooting this issue. 

### What is the expected behavior?

Deluge responds only to learned or assigned midi channel. Current behaviour is receive on all channels. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0-Nightly-33B625E

### If possible provide the steps to reproduce the issue and upload additional media:

Enter midi follow channel select submenu. Hold learn, send midi note from desired channel
Send midi note of any other channel > Deluge responds as if learned channel was sent.

seangoodvibes, 2024-02-03:
@schtaf would it be possible for you to come join us on the discord to help troubleshoot further?

I asked a few others to confirm if they have this problem and they said no.

Do you have other midi follow channel's learned? There's 3 channel's A/B/C. If you have more than one channel learned than the notes will make it through.

If all 3 channel's are unlearned no notes should pass through.

And if only 1 channel is learned, only notes on that channel should be allowed through.

There's nothing in the code to suggest that notes on unlearned channel's would make it through.

schtaf, 2024-02-04:
Would be happy to help and get help further. I’ll simply my setup to the basics and work from there. I am happy to hear it may be user error. The ability to turn these features off in a simple manner might be beneficial.  Will be o ln in a few hours.

Thanks!!!

Stefan

seangoodvibes, 2024-02-04:
> Would be happy to help and get help further. I’ll simply my setup to the basics and work from there. I am happy to hear it may be user error. The ability to turn these features off in a simple manner might be beneficial.  Will be o ln in a few hours.
> 
> 
> 
> Thanks!!!
> 
> 
> 
> Stefan
> 
> 
> From: Sean Ditny ***@***.***>
> 
> Sent: Saturday, February 3, 2024 6:49:26 PM
> 
> To: SynthstromAudible/DelugeFirmware ***@***.***>
> 
> Cc: schtaf ***@***.***>; Mention ***@***.***>
> 
> 
> 
> 
> 
> 
> @schtaf<https://github.com/schtaf> would it be possible for you to come join us on the discord to help troubleshoot further?
> 
> 
> 
> I asked a few others to confirm if they have this problem and they said no.
> 
> 
> 
> Do you have other midi follow channel's learned? There's 3 channel's A/B/C. If you have more than one channel learned than the notes will make it through.
> 
> 
> 
> If all 3 channel's are unlearned no notes should pass through.
> 
> 
> 
> And if only 1 channel is learned, only notes on that channel should be allowed through.
> 
> 
> 
> There's nothing in the code to suggest that notes on unlearned channel's would make it through.

Turning the features off is pretty simple:

- you can use shift + learn in the channel menu to unlearn the channel
- or use select encoder to scroll to unassigned

If channels are unassigned, midi follow ceases to work

m-m-adams, 2024-02-04:
I think it might be your MPE settings. If you have the device set with 16 channels assigned to either upper or lower MPE zones the problem you're describing would happen. That setting is in midi->devices->(your device name)->mpe->input->upper/lower and set them all to zero

schtaf, 2024-02-08:
Hi all,

Thank you for the replies and helpful suggestions so far. I think there is a large amount of user error here. I’ve tried several things:

unlearned all midi follow channel settings, including feedback 

Setup 1 keystep > deluge via din
learn channel A, resulting settings

use any device
DIN ports
Channel 10

Everything behaves as expected

setup 2
Deluge and keystep connected via USB to ipad . Midi routing is done by aum.
Learn channel A, resulting settings

use any device
Upstream USB port 1
MPE lower zone

This results in the activity where deluge responds to multiple channels

if after learning the channel, if I manually select the channel, D deluge does not respond to incoming messages. 

I can use AUM to troubleshoot and develop workarounds and have used midiscope to look for duplicate messages etc and have found none. 

Thanks for any help or advice

m-m-adams, 2024-02-09:
Perfect - the fix is to go to settings->midi->devices->usb upstream port 1->mpe->input and set both upper and lower zones to zero. Right now it is treating the input as an MPE zone. This can happen if you've ever sent notes on multiple channels while learning midi, deluge interprets that as you wanting to set that range of channels as an MPE zone

closed 2024-02-09 by m-m-adams

schtaf, 2024-02-09:
Will try and confirm back here tonight. What a fantastic community thank you all.