SynthstromAudible/DelugeFirmware #1204: Crash - kit mode with keyboard - 7seg

jay-poq, 2024-02-10:
### Please describe the problem:

versions:  
- c1.0.1 (amadeus)
-  v1_1_0-nightly+2024_02_10-9d1cbcb.

repeated and consistent crash (with high pitch sound) when using a kit in keyboard mode with the pads resized to a smaller level than default.  kit parts are built from two 10 second-ish wav files with different trigger points.   there are about 10 samples inthe kit. 

### What is the expected behavior?

should not crash 

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

- load track that was saved in community firmware 1.0.1
- navigate to kit track with 10-ish samples (vocal)
- switch to keyboard mode and resize pads to smaller

m-m-adams, 2024-02-11:
I can't reproduce this, it's crash free for me down to the minimum pad size of 1. Could you make a collect all save of the song and share it please?

jay-poq, 2024-02-11:
thanks Mark!  will try a few more replication steps then will either share the song or update on further findings.

jay-poq, 2024-02-15:
ok, so have spent a bit of time trying to reproduce, and i do get the crash consistently.
here are the factors:
* back to firmware 1.0.1
* thought it might have been an issue with audio corruption, so resampled the wav file in the deluge and built a new kit from the sample
* resaved the song under the new firmware
* used the new kit (#2 in the song file)
* without keyboard view - couldn't replicate the crash
* in keyboard view, with the drum pads resized to 3x3 it consitently crashed (though it took longer to crash after i deleted a few tracks).

weirdly, the song file seems to still reference the original sample, not the one in the collected folder?   that has been attached for reference. 
[newWrench.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14303292/newWrench.zip)

m-m-adams, 2024-03-03:
Any updates on this issue?

commit 858ca7f7 (2024-04-13): oled: fix popup border clamping (#1705)

This might fix #1204. The sample song from that bug has ultra-long sample names
which result in maximum size popups. Given the other symptoms in the bug I
think some sort of memory corruption is at fault -- this at least constraints
the popup to the renderable region. Other fixes we've put in over the last few
months hopefully got the rest of the corruption if there was something else
going wrong.
PR #1705 (2024-04-13): oled: fix popup border clamping
This might fix #1204. The sample song from that bug has ultra-long sample names which result in maximum size popups. Given the other symptoms in the bug I think some sort of memory corruption is at fault -- this at least constraints the popup to the renderable region. Other fixes we've put in over the last few months hopefully got the rest of the corruption if there was something else going wrong.
PR #1720 (2024-04-13): oled: fix popup border clamping
This might fix #1204. The sample song from that bug has ultra-long sample names which result in maximum size popups. Given the other symptoms in the bug I think some sort of memory corruption is at fault -- this at least constraints the popup to the renderable region. Other fixes we've put in over the last few months hopefully got the rest of the corruption if there was something else going wrong.
commit 33c47e26 (2024-04-13): oled: fix popup border clamping (#1705)

This might fix #1204. The sample song from that bug has ultra-long sample names
which result in maximum size popups. Given the other symptoms in the bug I
think some sort of memory corruption is at fault -- this at least constraints
the popup to the renderable region. Other fixes we've put in over the last few
months hopefully got the rest of the corruption if there was something else
going wrong.
closed 2024-04-13 by m-m-adams