SynthstromAudible/DelugeFirmware #1209: [Automation View] Source automation Not Preserved When Copy Pasting Automation

sdesign75, 2024-02-11:
### Please describe the problem:

I’ve found an issue when copy/paste automation. Something is happening to the page that you copy from after you paste the automation to the next page. The first automation point disappears. Also, the whole first page seems to not respond to the automation when first played. Also happens when copy pasting automation between different parameters e.g. master level to filter cutoff.

Note: On c1.0.1, the first automation point disappears, but once you redraw this the pattern/automation plays back fine.

### What is the expected behavior?

Original data is preserved when copy pasting automation

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.1.0-NIGHTLY-9D1CBCB

### If possible provide the steps to reproduce the issue and upload additional media:

1 - Empty synth clip
2- Double All, Double All to create 4 bar pattern
3- Add a long sustained note on each bar
4 - Press Clip to enter Automation View
5 - Select Master Level
6 - Press and <> to zoom to 16th notes
7 - Create automation on page 1 like below, where X is 100% and _ is 0%

X_X_XXX_X_X_XXX

8 - Hold Learn/Copy and press Gold Encoder to copy the automation
9 - Use <> to move to second page
10 - Shift+Learn/Copy and push Gold encoder to paste the automation
11- Now use <> to go back to page one. The first part of the automation is missing. It’s set at 0%
12 - Now paste automation into pages 3 and 4
13 - Reenter automation on first step of page 1. Set to 100%.
14 - Press Play. The first page of the automation is not heard. It seems to be set at 0% throughout

sapphire-arches, 2024-04-03:
Still happens as of 64d750cd174953ee770dd460b3d0feb1c048e98d (today's top-of-tree). I suspect we're not handling note array reallocation correctly in the copy/paste logic.

sapphire-arches, 2024-04-09:
Using the regular clip view automation copy/paste shortcut has the same symptoms, so the bug is either in `AutoParam::paste` or in how automation view is setting up nodes. I'm inclined to believe it's the former over the latter.

PR #1901 (2024-05-12): Fix automation pasting
Fixes #1209.

This turned out to not actually be memory corruption, just logic wonky enough to look like it was. The new behavior should feel much more consistent, as it does its best to build a solid fence around the region being pasted in to and prevent the new start/end values from spilling in to adjacent regions.

The PR contains some bonus refactoring and doc additions.
commit d730340a (2024-05-12): Fix automation pasting (#1901)

* Remove goto in AutoParam::paste

* paramnode: c++ modernization

* automation: fix bugs in copy logic

Some cleanup for CopiedParamAutomation
 - the destructor was virtual for no reason (it has no subclasses or superclasses)
 - Fix a memory leak where we would blindly overwrite the nodes pointer
 - add documentation to the members (and for ParamNode)

Bug fixes:
 - restore subtraction of startPos in copy logic
 - set the pos of the inserted virtual first node to 0

* autoparam: fix restart node insertion logic

Fixes #1209

The previous behavior was sort of lo [... cut]
PR #1906 (2024-05-12): Fix automation pasting
Fixes #1209.

This turned out to not actually be memory corruption, just logic wonky enough to look like it was. The new behavior should feel much more consistent, as it does its best to build a solid fence around the region being pasted in to and prevent the new start/end values from spilling in to adjacent regions.

The PR contains some bonus refactoring and doc additions.
commit 4d6ce3df (2024-05-12): Fix automation pasting (#1901)

* Remove goto in AutoParam::paste

* paramnode: c++ modernization

* automation: fix bugs in copy logic

Some cleanup for CopiedParamAutomation
 - the destructor was virtual for no reason (it has no subclasses or superclasses)
 - Fix a memory leak where we would blindly overwrite the nodes pointer
 - add documentation to the members (and for ParamNode)

Bug fixes:
 - restore subtraction of startPos in copy logic
 - set the pos of the inserted virtual first node to 0

* autoparam: fix restart node insertion logic

Fixes #1209

The previous behavior was sort of lo [... cut]
closed 2024-05-12 by m-m-adams