SynthstromAudible/DelugeFirmware #1222: E339 Crash when creating new Kit and attempting to add a Synth to a Kit row

gobcmmelo, 2024-02-13:
### Please describe the problem:

E339 Crash when creating new Kit and attempting to add a Synth to a Kit row

### What is the expected behavior?

1. Hold blank Kit row Audition Pad and pressing Synth directs to Load Synth page.
2. Scroll to a Synth and pressing Select loads the Synth to the Kit row.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

C1.01

### If possible provide the steps to reproduce the issue and upload additional media:

The error appears at different times but is consistent in its appearance.

1. Hold blank Kit row Audition Pad and pressing Synth - Crash
OR Directs to Load Synth fine.
3. Scroll to a Synth and pressing Select - Crash

x3003, 2024-02-13:
On the 1.1.0 nightly, I don't get the error, but it just locks up the whole machine.
You don't even have to make a new kit, just press 'kit', then the audition pad and 'synth' together -> lockup....

m-m-adams, 2024-02-13:
Hey fyi this is in our release notes - https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/release_1_0_1

If you have no numerically named synths and you're still getting this crash definitely let me know - it's been particularly hard to pin down the cause of this

m-m-adams, 2024-02-13:
Disregard, this is a different crash fixed here - https://github.com/SynthstromAudible/DelugeFirmware/pull/1226

gobcmmelo, 2024-02-13:
> Hey fyi this is in our release notes - https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/release_1_0_1
> 
> If you have no numerically named synths and you're still getting this crash definitely let me know - it's been particularly hard to pin down the cause of this

I bet, it will reliably crash in this workflow but not always at the same point in the process. Does having numerically named synths mean any synth in the whole library that have a numeric name? Further does this mean synth names that are entirely numeric or include numbers in the name? Thanks!

m-m-adams, 2024-02-14:
Try the fix here: https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/7893970544/artifacts/1243153469

This should fix the immediate crash on pressing audition+synth, I noticed it's a different error than the one referred to in the release notes.

For numeric name issues I believe they can come up whenever a synth named like 123 or 123a is in the next couple synths to be scrolled through. I'm not certain though, and I haven't been able to get that specific crash consistently. If it happens it'll come up as error S001

PR #1226 (2024-02-14): check that drum exists before dereferencing
Fix #1222
closed 2024-02-14 by seangoodvibes