SynthstromAudible/DelugeFirmware #1239: Norns layout not highlighting incoming notes in nightly

Quixotic7, 2024-02-16:
### Please describe the problem:

Running nightly 20240216 and in norns macro mode https://github.com/SynthstromAudible/DelugeFirmware/pull/250 it is no longer highlighting the incoming notes from norns midigrid. 

It will work if I do a midi learn, but then the notes get sent back out on channel 16. 

I tested with Amadeus to sanity check, and it worked automatically without needing a midi learn. 

### What is the expected behavior?

Expected behaviour is to set clip to midi 16, enable keybaord view, switch to norns layout and see the pads light up with incoming grid notes from norns and only send out midi when I press a key on the Deluge. 

### Is there a relevant Pull request?

https://github.com/SynthstromAudible/DelugeFirmware/pull/250

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

20240216

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on Deluge, enable norns layout feature and note highlighting feature. Create midi clip, set to channel 16. enter norns layout. 

Deluge is powered via adapter. 

Norns is connected from usb-a to usb-b on Deluge. 

Norns midi is using Deluge 1 in midi device 1, midigrid mod is enabled. Tried with a handful of grid scripts including Awake.

n3ptun3s, 2024-02-16:
I tried it also recently with a nightly build.
Same result. 
I can send notes to the scripts, but the notes on the deluge do not get lit.

m-m-adams, 2024-02-22:
Fixed by #1257

closed 2024-02-22 by m-m-adams