SynthstromAudible/DelugeFirmware #1248: File Corrupted when Changing Preset

arealhumanbeing, 2024-02-17:
### Please describe the problem:

Message: File corrupted. 

That save is no longer able to be loaded. 

[image]

### What is the expected behavior?

Just a change in preset without corruption. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

C1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Made a white clip from Arranger Mode, kept it a synth. 
2. Went into it and entered a simple melody.
3. Turned select knob to choose new sample-based preset. 
4. File corrupted.

sapphire-arches, 2024-02-17:
If you still have it, the `song.XML` file that reproduces this bug would be very helpful. It might also be possible for us to recover most of the song.

The error is a generic consistency check failure after loading the song, so without the song XML it's unlikely we can fix this.

m-m-adams, 2024-03-03:
Seems to work fine - will reopen if you provide an example that crashes

closed 2024-03-03 by m-m-adams