SynthstromAudible/DelugeFirmware #1249: Unison pan-spread function for samples cauess clicks / pops

synthpatches, 2024-02-17:
### Please describe the problem:

When using a sample as an oscillator (or two samples), the unison-pan spread function (community feature) causes a lot of clicks and pops while playing. Crackles occur at the beginning of a pattern or when playing other short kick samples in the same time.

### What is the expected behavior?

No clicks / pops :) Currently, this is unfortunately a problem that almost always occurs when using pan spread for samples.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

C1.0.1

M0R172, 2024-02-18:
I tried to reproduce this.

It seems to me that the problem is not the stereo spread, but that I generally get clipping (also with stereo spread 0) on a sample oscillator when using high unison numbers OR playing the voice with high polyphony. It seems that, for a full amplitude sample, clipping occurs roughly when (unison*number of notes played) > 8.

So I guess there is some missing amplitude adjustment somewhere in the voice code - which I don't understand well enough to try and fix myself...

bfredl, 2024-02-18:
> It seems that, for a full amplitude sample, clipping occurs roughly when (unison*number of notes played) > 8.

Since multiple notes contribute to it, does it help to just lower the global volume of the voice (MASTER LEVEL param)? we can try to retune it but it might be tricky as people have samples of different prior loudness.

M0R172, 2024-02-18:
Just tested, and yes, reducing the Synth's "master level" actually makes the clipping go away, but I think the user should not have to do this manually - would be interesting if the problem exists in official as well...

As for samples of different loudness, maybe so, but there is a well-defined maximum amplitude that a sample can have, so it should be possible to safely avoid clipping. If users have samples that are too quiet, maybe one could introduce a "normalize" feature somewhere, or some additional adjustable volume boost for sample voices - but that is a completely different topic...

M0R172, 2024-02-18:
BTW best way I found to test/reproduce this, is 
- make new synth (shift synth)
- set osc1 to a melodic sample 
- use some LPF to better hear the clipping overtones, 
- crank up unison to 8 
- play several notes at once

synthpatches, 2024-02-19:
Here you can download the test song with a sample-based patch which causes the clipping problem. Just move the song and wave files into sd card and play the song: https://synth-patches.com/test.zip

ok-reza, 2024-02-20:
hey @synthpatches i'm getting a "[404] File Not Found : Plik nie istnieje" error with that link

M0R172, 2024-03-09:
OK, now I tried it with this file, the issue is a separate one from the clipping that I had.

This here is *definitely not* related to unison or pan spread! (someone with mod rights could change the title...)

Proof: Reduce unison to 1 - clicks and pops are still there.

@m-m-adams : I strongly suspect this is culling going wrong - so maybe you could have a look into it... the sound in question is a 6.7-second sample played on a lot of notes, and the problem goes away when decreasing the envelope release - actually the maximum release time before it starts popping is clearly inversely related to the unison number...

m-m-adams, 2024-03-09:
Not sure it's misbehaving - this synth is very heavy with the long stereo sample+delay and the sound is the result of hard culling

Community is able to render 13 voices of "synth 76" while official can only maintain 10. Unfortunately with the high release in this song the result is that community hard culls multiple voices when the kick comes in while official is soft culling the synth throughout which is less clicky.

m-m-adams, 2024-03-09:
Nudging the kick one tick right solves the problem entirely

M0R172, 2024-03-09:
Hard cull means zero fade-out, hence the clicks?

m-m-adams, 2024-03-09:
Yeah - there's three types of culls 

1. Soft cull, which starts a very short release roughly twice as fast as a release of 1 
2. Immediate cull, which releases over the next render window and can click a little bit if unlucky. This is only in community 
3. Hard cull, which immediately deallocates the voice with no further rendering

Hard culls click but they're necessary where we can't be sure that the next render will finish on time, since that will cause even bigger clicks

m-m-adams, 2024-03-10:
@synthpatches you can experiment with the build from here - there's some side effects still but it automates the process of nudging notes to avoid clicks, which helps with this song

https://github.com/SynthstromAudible/DelugeFirmware/pull/1453

sapphire-arches, 2024-04-03:
I looked in to this and it turns out we already have unison count compensation, and as near as I can tell it applies to unison spread voices as well. The caveat is that the compensation scales the peak-to-peak value by `1 / sqrt(numUnison)` rather than the naive `1 / numUnison` -- this gives accurate scaling in energy rather than peak-to-peak value. Scaling by energy is probably about as close to perceptually uniform as we can get without doing some really gnarly analysis on the sound.

Since we already have compensation and it's probably about as good as we can do, I'm going to close this for now -- please feel free to reopen or otherewise discuss alternative compensation schemes though!

closed 2024-04-03 by sapphire-arches