SynthstromAudible/DelugeFirmware #1259: Arpeggiator tweaks

m-m-adams, 2024-02-19:
- Ratchet doesn't work with playback stopped
- Ratchet doesn't work on MIDI notes
- Arpeggiator menu classes are redundant 

@soymonitus

soymonitus, 2024-02-19:
What is "arpegiator menus redundant"?

soymonitus, 2024-02-20:
Ohh I know what you mean. Gate was already like that from before I started coding, and it is because the Params are different for synth and for midi clips. There is a different version of the menu items for each. It seems that Params can't have the same behaviour when for midi clips, they save the value differently. I just mimicked Gate for my new params.

m-m-adams, 2024-02-20:
No, the gate and sequence length classes are identical. There is no need for them to be seperate classes. 

[image]

Same situation for ratchet and ratchet probability

soymonitus, 2024-02-20:
Ohh i see

soymonitus, 2024-02-20:
Redundant arpeggiator menu classes is fixed here:
https://github.com/SynthstromAudible/DelugeFirmware/pull/1271

closed 2024-02-21 by sapphire-arches