SynthstromAudible/DelugeFirmware #1264: Voice drops that don’t occur on official

Icoustic, 2024-02-19:
### Please describe the problem:

Here’s another of my projects made on official that have a lot of voice drops even on one of the latest nightly builds, feb 19. All the active tracks in arranger play normally on official firmware

Collect-zip: https://we.tl/t-0yPHzgDbVi

### What is the expected behavior?

No voice drops, at least very minimal 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### If possible provide the steps to reproduce the issue and upload additional media:

Play the song, and voices drop, such as the respirator sample, synth pad etc.

m-m-adams, 2024-02-20:
Great, thanks for sharing! This actually culls a lot of voices in both official and community - on the whole community is culling less, but the times/voices it chooses to cull are much worse. I'm going to stew a bit on how to fix this!

m-m-adams, 2024-02-20:
Any chance you have a stereo version of this song? It's a bit uncomfortable to listen to in headphones and I'll be hearing it on repeat this week

closed 2024-02-20 by seangoodvibes

Icoustic, 2024-02-20:
> Any chance you have a stereo version of this song? It's a bit uncomfortable to listen to in headphones and I'll be hearing it on repeat this week

Noo I work in dual mono send/ return setup, sorry. You can pan stuff back to center if you must 😖

PR #1269 (2024-02-20): use correct limit for solicit via culling
Solicit via culling accidentally compared to the direnessThreshold (used for lowering FM/filter quality) and not the numSamplesLimit, causing it to steal voices unnecessarily 

Soft culling accidentally used numSamplesOverLimit instead of smoothedSamplesOverLimit - as the render routine will be immediately recalled in dire situations anyway, this allows more accurate estimation of performance without the risk of falling short on needed samples

Fix #1264 - although I think this song should be playable with no culling whatsoever with more tweaks so I'll [... cut]