SynthstromAudible/DelugeFirmware #1284: Samples Still Being Dropped in Beta 1.1.0

chewabledrapery, 2024-02-21:
### Please describe the problem:

This is a follow up from #804 and #1225.

With projects that use samples i am still getting random samples being dropped/not played and it happens on relatively simple projects. 

Example2 project and a video of the behaviour on Facebook linked below 

https://www.facebook.com/groups/synthstromopen/permalink/3755557178097177/?mibextid=c7yyfP

Thanks all.

### What is the expected behavior?

With only 3 or 4 samples being played I wouldn't expect this behaviour and it doesn't do it on the official fw.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

C1.1.0

### If possible provide the steps to reproduce the issue and upload additional media:

[Example2.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14356062/Example2.zip)

chewabledrapery, 2024-02-21:
Changing the polyphony of the kick from AUTO to POLYPHONIC improves things somewhat but changing patterns then still creates spurious dropouts, I don't know if this helps shed more ligh? Maybe the issue lies with polyphony/chokes rather than voices...

m-m-adams, 2024-02-21:
Official performance on this song for reference:

[image]

and community - note that the "internal memory full" diagnostic was added by us, it's just not logged by official.
[image]

Both are culling - official is culling very frequently (1-2 times per beat) and constantly remaining in the high direness state. Community is culling rarely (once per bar or sometimes not at all) and dropping into lower direness, but the times that community culls are much worse for your song since it tends to mean dropping the kick. 

What you can do is set the kick priority to high so that it's never the voice that gets dropped. Because your problem comes from filling the internal memory, it would probably also help to delete unused kit rows in your kits - kit8 in particular is probably using a ton of ram. While it's not ideal to require tweaks to your old songs, I don't see an immediate way to fix this without negatively impacting performance in other cases

chewabledrapery, 2024-02-21:
So fundamentally this is due to using a large sample and slicing it thus using too much memory?

As Raul Monitus suggested on my Facebook post I'll try the Kit + Shift + Delete to batch delete unused kit samples and also just be less lazy and properly audit what and how many samples I'm using in a kit. 

I've learned a lot here, guess I just wasn't really understanding that although the project "wasn't big" my overuse of too many samples was exceeding reasonable capacity. 

Thanks for taking the time to look into this and making the cause clear. 

Cheers.

chewabledrapery, 2024-02-21:
Closing as the issue seems related to overuse/mismanagement of samples in a project thus stretching the Deluge's finite resources too far.

closed 2024-02-21 by chewabledrapery