SynthstromAudible/DelugeFirmware #1285: E451 on song load

Balistix55, 2024-02-21:
### Please describe the problem:

I have an E451 on song load. This error only appears in the case of the beta version. I tried the same song on Amadeus, on 4.1.4, on nightly from 2nd February and there song loads ok with no error. In the case of the beta version, the error manifests itself in the arranger, where instead of several consecutive mapped midi channels (1-6), only channel 1 is shown on all six rows. Also, all clips disappear for these midi rows. Midi channels 2-6 appear in the arranger in the rows below the other instruments and samples. When I look at the midi channel clips, they have completely messed up CC mapping.

[Cs1X Beast 9.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14364303/Cs1X.Beast.9.zip)

### What is the expected behavior?

Load a song without E451

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

Beta

m-m-adams, 2024-03-03:
This no longer crashes but still doesn't work

Balistix55, 2024-03-03:
Do you have any idea, what causes it? Only midi clips are affected. I have 2 songs (in both cases with midi clips / channels) with this same error behaviour.

m-m-adams, 2024-03-03:
No not yet - hopefully we'll figure it out soon!

Balistix55, 2024-03-03:
Wish you luck 🤞🏻

sapphire-arches, 2024-04-03:
I can no longer reproduce this crash, but it appears the song zip above is missing most of the samples -- if something is going wrong in the sample loader and corrupting memory this may still reproduce.

@Balistix55, can you try a recent beta build on your device? I'm going to close this for now but if you still can't load that song file on your device please reopen with a zip including the folder generated by the collect-all save.

Balistix55, 2024-04-03:
Yeah, there was no problem with samples, but with midi channels, that's why I did not collect all song with included samples, but only song data. If you focus on midi channels and clips created with them, there is some strangeness in arranger mode. You can compare arranger in official FW with arranger in community FW. In midi clips there are CCs mapped and also recorded automation. There are 6 midi channels. On official FW everything works well. On community there is an error e451 on song load and if you look in arranger, you can see that instead of midi channels (instruments) 1,2,3,4... appears 5 or 6 times midi channel 1. Other midi channels (2,3,4...) are listed at the bottom of arranger instrument list. Also if you go into midi clips, CC params are stuck in the middle without automation. I will look into song xml file and try to find some clue and then maybe rebuild the song ;)

Balistix55, 2024-04-03:
Maybe I can save a song with only a few channels and instruments and collect all for better overview? I will do it today after work ok?

sapphire-arches, 2024-04-03:
no rush, I don't know if I'll have much time in the next few days. I think the best thing to do would be saving and reloading the project after each element until you have a project that doesn't save correctly.

sapphire-arches, 2024-04-04:
E451 means that a clip got associated with the wrong "output" (internal name for instrument-like things). I'm pretty sure it can only happen if there's a bug that corrupts memory.

m-m-adams, 2024-04-07:
Bad news - this still attempts to generate E451 but the popup doesn't appear for some reason.

Balistix55, 2024-04-07:
@m-m-adams what about if I try to compare this xml song file with another one (not the same, made with latest beta or nightly, also with midi cannels, clips, CCs, automation)? There has to be some nonsense in file ;) This issue appeared with one of the nightly that day or few day before I made this issue. Question is if there were any changes in xml file structure?

Balistix55, 2024-04-07:
If it will not be succesful I can remake the song.

closed 2024-04-08 by sapphire-arches

m-m-adams, 2024-04-08:
We got it fixed, it'll work on tomorrows builds

PR #1684 (2024-04-08): Bugfix/midi load error arranger
Fix #1285 , an incorrect tag name was set when internal was added as a midi destination

Stops the song loader from creating fake midi/cv clips if the song loads incorrectly, making errors like this more obvious. 

Also stops the session/arranger display from killing error messages during loading to ensure the error message is shown
PR #1686 (2024-04-08): Bugfix/midi load error arranger
Fix #1285 , an incorrect tag name was set when internal was added as a midi destination

Stops the song loader from creating fake midi/cv clips if the song loads incorrectly, making errors like this more obvious. 

Also stops the session/arranger display from killing error messages during loading to ensure the error message is shown