SynthstromAudible/DelugeFirmware #1296: Massive Voice stealing in drum kits with Beta firmware.

highsiderr, 2024-02-22:
### Please describe the problem:

Massive Voice stealing with new Beta Version

### What is the expected behavior?

If I load the track with the 1.0.1 firmware, no voices are stolen. 
With the new beta, voices are stolen in the drum kit.
[songs final.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/14373013/songs.final.zip)

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

deluge-c1_1_0

m-m-adams, 2024-02-22:
Confirmed - on last release this is hard culling occasionally, on current beta it's soft culling frequently. 

I'm going to look at tweaking the cull thresholds - with the changes to the cull system I think we can push them a bit higher now

Thanks for sharing the song! It's a fun listen

highsiderr, 2024-02-23:
THX @m-m-adams

PR #1299 (2024-02-23): tweak thresholds
With soft culling now working reliably we can push the thresholds up a bit more

Also set a minimum voice number, and don't cull if we're above that

Fix #1296 

Possible chance of overshooting rendering, but I'm not noticing it in my amassed collection of problem songs
closed 2024-02-23 by stellar-aria