SynthstromAudible/DelugeFirmware #1307: Request: make Pressure on MPE keys update the velocity of the arpeggiated notes

soymonitus, 2024-02-23:
### Please describe the problem:

It would be nicer if the pressure expression parameter from an MPE controller, when holding notes with your fingers, could update the velocity sent by the post-arpeggiator notes. Right now if you initially play softly the chord to be arpeggiated, it doesn't matter if later you increase pressure on the keys, all post-arp notes will sound super quiet.

### What is the expected behavior?

Notes get higher velocity as you increase pressure on the keys of your MPE controller

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

soymonitus, 2024-02-23:
Not sure if it is considered a bug or a nice-to-have. Anyways I think it would be nicer, if your controller is MPE capable, that the arpeggiated notes generated by the deluge express better you current finger positions when a new note is sent out.

m-m-adams, 2024-02-23:
I agree, it works this way on the linnstrument arpeggiator and it's quite nice

closed 2024-02-25 by m-m-adams