SynthstromAudible/DelugeFirmware #1308: Enter button in Keyboard mode unresponsive

Tacosphere, 2024-02-23:
### Please describe the problem:

Loading new kit or synth with press of New button brings me to the keyboard mode. 
I type in the name I search for. No search options appear, pressing enter buttons (green) with enter symbol doesn't response. Pressing shift plus enterbutton neither responds.
I am asking whether I missed something... Could be me but it seems strange to me that no search results appear.
Toggling the knob next to the oled just offers me presets starting from 001 .
Am I doing somthing wrong here or is there a bug with the enter buttons?

### What is the expected behavior?

searching for names in synth and kit mode should present found results dynamically and after pressing Enter sound should be loaded. This button (2green) don't respond or any search result is presented. Even simple ones like drum or set or perc. e.g..

### Is there a relevant Pull request?

N/A

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

C1.0.1

### If possible provide the steps to reproduce the issue and upload additional media:

yes

m-m-adams, 2024-02-23:
I can't reproduce this - it's working fine on my deluge. Can you take a video?

ok-reza, 2024-02-23:
works fine on my end. keep in mind searching is only for presets and folder names in that folder, if you are searching for a preset name that is within another folder, it won't show up. searching has to be an exact match, so i normally type a letter or two at a time.

closed 2024-02-28 by m-m-adams